evil-mad/robopaint #203: 64bit or 32bit OSX Build?

techninja, 2015-01-25:
On the latest node-webkit (now renamed to nw.js) builds, now up to 0.11.6, there are 32 and 64bit builds for mac (actually, all the systems). There are MANY Windows machines not quite up to 64 bit (though the number dwindles as all Win8 will be x64). I then wondered.. should we make the next release x64 for Mac? what's the split?

According to [this page](http://support.apple.com/en-us/HT3696), apparently all Macs starting after core 2 duo (some nearly 10 yrs ago) were x64. I think...?

Using this info, I'm going to compile the native modules to x64 and use the x64 binaries only for 0.9.5. I won't do this for Linux or windows (unless I hear people want em). Better to support the most people with the best we can. I've placed this in the 0.9.7 milestone on the hope that we can have some kind of consensus on this before then :hammer:

oskay, 2015-01-27:
A little bit subtle.  
http://apple.stackexchange.com/questions/99640/how-old-are-macs-that-cannot-run-64-bit-applications

techninja, 2015-02-17:
Doing some cleanup. I've referenced this in the release notes, so in the off chance we come across anyone in need of a 32bit release we can make it, but there should be little _need_ for it apparently.

closed 2015-02-17 by techninja