evil-mad/robopaint #204: Roll 0.9.7 feature/bugfix release

techninja, 2015-01-29:
_This issue is a placeholder for the work and discussion behind the next release and should act as a tracker for any release delays or important points about the build not already ticketed._

This release is intended to be the last planned release before 1.0.0. The goals of v1 are to ensure a fast, comprehensive, and mostly bug free environment to use at least two drawing robots. Everything is basically on track for this, though "comprehensive" is a bit up in the air, I'd interpret it to allow for more drawing modes by default, and especially one for raster image conversion.

For 0.9.7, we'll be focusing on:
* #130 - EggBot support: Over the course of dev for 0.9.5, it was found that the proper way to do this involved adding a bit too much, so while the branch was started, lots more work will be done. I certainly hope to make this one of the main focus', apart from other features/fixes as they come, this one is going to drive a lot of user experience and is therefore pretty important.
* #120 - Revamped quickload: This is somethign I've been wanting to do forever, especially to replace the really bad looking quickload menu I wrote myself one afternoon 2 years ago.
* #118 - Sub path painting: @gwygonik had originally submitted this one, and honestly it's a great idea, I just need to buckle down and find out the methodology and ensure it applies cleanly. I suspect with a few extra settings (min/max reink after/before path switch) with will perform quite well and certainly better than the existing non-path-aware algorithm.
* #34 - Automated updating: I'm voting this most likely to fall off the plate for this release. It would be awesome to have automated updating, but it's a whole new system to manage, for all three OS'. It's an updater, or an annoying message with a link :P.

techninja, 2015-02-15:
0.9.7 progress is going **_terribly_** what with FRC in full swing, #130 still having open questions keeping work from simply being head down implementation, and my day job/weekends have been unrelentingly busy. Not to mention trips to various teaching conferences are up and coming and are sure to sap any extra free time.

I've moved 0.9.7 release date to March 10th, and even that doesn't feel concrete unless we can get the implementation details for #130 wrapped ASAP, and I can get a move on on the other important pieces of this release.

The very first WaterColorBot was born 2 years ago this next week. I'd hoped to get RoboPaint v1 out by then, but that's just not in the cards. Oh well.

# Happy Birthday WaterColorBot!

[image]

oskay, 2015-02-15:
Renewing WaterColorBot.com domain name today. :D

oskay, 2015-02-16:
FRC in slightly-less than full swing soon. Should go back to our regular amount of not-having-free-time.

techninja, 2015-02-17:
Wonderful to hear. Hopefully we can batten down the hatches and squish together some semblance of a respectable release while also prepping some "Ohh & Ahh" features for RoboGames 2015. I've got some ideas (that require work and testing) that should fit well with the system... Just can't leave this big tickets alone for too long, they grow fur, turn green and get a life of their own. :mask:

techninja, 2015-03-12:
As should come as no surprise, 0.9.7 is yet again delayed, but not without cause:
* I finally met the "away team" of @rogerfachini & @docprofsky (and their Dads!) in person at an event in LA to at least partially discuss general plans and hack away a bit on things. Meeting was fruitful, enjoyable, and sparked the initiation to a grand plan culminating in their enthusiasm to make something "awesome" and represent at RoboGames 2015. We're registered and raring to go, it's going to be awesome.
* @makersylvia uncovered some **_literal_** showstopper bugs while showing the bot off to a number hundreds of educators in Chicago (#209), I suspect I'll triage the report and figure out exactly what the issues are and get them fixed for this release, or very likely as a 0.9.6 bugfix considering the likely number of new WaterColorBot owners :)

### General progress/status:
* #130 is slowly getting there. The changes to the user interface to fully and cleanly support different bots are slow going mainly because of my lack of clean idea of how to implement everything well together. Years of being told what a good plan is and implementing it have stunted my ability to do everything quickly... Oh wait, I was never any good at that :P
* None of the other tickets have really been looked at or worked on. The guys have their RoboGames projects to improve, and I've got the latter ticket, and daily life.

I'm expecting to get a weekend or so to push through and finish 130, then figure out the rest of what's needed, but I suspect the bugfix release is going to need to take priority considering the performance issues I've been seeing.

docprofsky, 2016-09-28:
Seeing where RoboPaint is at and going we should skip this release and just do a 2.0.0 release.

techninja, 2016-10-03:
Yea. Closing.

closed 2016-10-03 by techninja