evil-mad/robopaint #205: Add SVG text handling ability

oskay, 2015-02-05:
If you save an SVG document (say, from Inkscape), it will open and render properly in RoboPaint (edit mode) but does not appear in the Print mode.

Separately, it would be nice to add the ability to add text directly in the edit mode.

techninja, 2015-02-05:
I've toyed with this a bunch, the problem kind of comes down to fonts and paths. The ONLY thing getting traced in RP right now are paths. If it's not a path, it's tossed out. Text is stored as the actual text and rendered not as paths, but more directly based on their style. Sure, you can convert them to paths, but being able to actually print them and have a selection of single path fonts is the trick.

Likely what is needed is conversion of the single line engraver fonts into SVG fonts, which I believe are supported by Method-Draw. That or we build out the example text render mode a bit into a method-draw plugin. OR... we support all the fonts and use the canvas SVG text render plugins available to actually render strokes/fills. The latter being the most difficult.

oskay, 2015-02-05:
In wcb-ink, we also ignore anything that isn't a path, but produce a message to the user when it encounters something that cannot be drawn.  

It seems that since method-draw can render the fonts, it must be possible (somehow) to render them into strokes and/or fills. 

Got a little stuck this week trying to explain how to paint text with WCB, so thought that there should be an issue to discuss these things.

techninja, 2015-02-17:
Well.. If we can SEE it on the canvas, we can use visual "path" checking for almost anything to fill, but we can't really trace/stroke it directly unless we can tap into the renderer. Apparently even this isn't impossible with the right library.

Interesting links:
* [vectorize-text node lib](https://www.npmjs.com/package/vectorize-text)
* [Paper.print() render text to SVG paths... apparently)[https://code.google.com/p/svg-edit/issues/detail?id=633#c4]

oskay, 2015-02-17:
"even this isn't impossible"  :D

techninja, 2015-09-23:
I'm not sure if it's enough, but the "example" mode for text rendering via HersheyText with live update does a very good job rendering text, though it's quite limited in font. @oskay if you think this mode is up to snuff (or at least has the basics of what we're after) in the latest build, go ahead and close this ticket.