evil-mad/robopaint #207: CALL ALL PEEPS!: 2 yr anniversary WCB User Poll Questions

techninja, 2015-02-17:
As inspired by @RI0's quest for a wireless WCB ecosystem, I thought it would be prudent to take our active users here on the Github Robopaint issue queue, to pull together a list of questions that could be put together on a survey for all WCB users. We're really missing a lot of information about how they use the hardware and software, and it would be great to find out what crazy idea's we've come up with they would find most interesting/useful. **_Calling all posters!_**

@RI0 @docprofsky @rogerfachini (and anyone else)

When we're done, we'll probably regroup them into something orderly and 

### Format: 
On the first line, put the type of question in square brackets, 2nd line put your question, then on the next lines, put the multiple choice options (if any) for the question.

**Types of Questions:**
 0. [%TYPE Rating] - User Rates on a scale of 1-5 their level of %TYPE, which could be "Satisfaction", "Happiness", or "Usability", "Strongly" (or whatever), with the options appearing as "Very strongly", "Somewhat Strongly", etc.
 0. [Multi] - User selects from the options given. If one of the options is "other", we'll just assume they can type something in at that point
 0. [Sinlge] - User selects _**only one**_ of the options given. If one of the options is "other", we'll just assume they can type something in at that point
 0. [Comments] - Users will just be given a form field to type out their answers

techninja, 2015-02-17:
[Multi]
What software to you most often use to control your WaterColorBot?
* RoboPaint (Desktop app)
* RoboPaint RT
* Inkscape Plugin
* Other

["Simple" Rating]
How easy is it for you to use your WaterColorBot?

techninja, 2015-02-17:
[Multi]
What operating system(s) to you most often use to drive your WaterColorBot?
* Windows 7+
* Windows XP
* Mac OSX > 10.6
* Mac OSX < 10.6
* Linux Desktop/Laptop
* Embedded Linux
* Other (or more details, please specifiy)

techninja, 2015-02-17:
[Single]
Have you, or do you plan to use your WaterColorBot in a classroom setting?
* Yes
* No
* I'm not sure how I could make it work
* Other

[Single]
Would you benefit or be interested in the ability for users on separate workstations to be able to create art, submit it, have it be queued, then painted/managed at a single WaterColorBot?
* No
* Yes
* Absolutely! This changes everything!

techninja, 2015-02-17:
["Important" Rating] {all the ones in this comment}
How would you rate the following in importance in the program you use to control your WaterColorBot?
* Program stability?
* Extra features in painting/programmatic art creation?
* Wireless access?
* Extra control/fine tuning art output/fill styles?
* Access from a mobile device?
* Beautiful/Usable and clear to understand user interface?
* Accessibility for the disabled?
* Ability to save/export any procedural content?

techninja, 2015-02-17:
["Important" Rating] {all the ones in this comment}
How would you rate the following art creation/output abilities within RoboPaint in importance (or interest of possible future use)?
* Import standard photographs?
* Import external SVG images?
* Set text with custom fonts?
* Input low level painting commands via scratch/snap block programming languages?
* Preloaded & tested images for demo or recoloring?

techninja, 2015-02-17:
[Multi]
What would you consider yourself/the users of your WaterColorBot most closely match in the following list?
* Student
* Maker
* Artist
* Hardware developer/hacker
* Software developer/hacker
* Happy mutant
* Designer
* Tinkerer
* General Hoopy Frood
* Other

[Single]
Is your WaterColorBot part of a hackerspace/fablab/maker place?
* Yes
* No
* Sometimes

[Single]
Did you know that there's a community of developers looking for help on the Open Source software projects that make the WaterColorBot awesome?
* I'm too busy painting to notice.
* No way! That's so cool. I should totally go tell them what I like and don't like!
* Yes, I help them whenever I can.
* I plan on buying them coffee and pizza when I see them at Maker Faire

rogerfachini, 2015-02-17:
[Comments]
What abilities could RoboPaint benefit from that it does not currently have? 

[Multi]
What language(s) do you/the user of RoboPaint prefer to read in? {this could probably be phrased better}
 - English
 - Spanish
 - French
 - ... {More common languages}
 - Other

docprofsky, 2015-02-17:
[Comments]
If you have a problem with RoboPaint what do you do to try to fix it or have it fixed?

["Agree" Rating]
- Limitations in RoboPaint limit what I can do with my WaterColorBot.
- The user interface feels cluttered and hard to use.

[Single]
Have you experienced issues loading SVGs into RoboPaint to draw?
- Yes
- No

docprofsky, 2015-02-26:
@techninja: 
Are we done with these questions? When will they be sent to WCB users?

techninja, 2015-03-03:
We need permission and verification from @oskay on these, and then he will do the sending. Come to think of it I never got word from him that we could/would even do this. Hrmm.

oskay, 2015-03-03:
Do you have a proposed method of contacting WCB users?

techninja, 2015-03-03:
I'm left to assume you have a list of customers and email addresses of who you've sent a bot to. We could also use the kickstarter list but, I'm not sure what the terms of use are. If both of these are not options, then we're out of luck.

docprofsky, 2016-08-02:
[Multi]
What robots do you use with RoboPaint?
- WaterColorBot
- AxiDraw
- EggBot
- Other