evil-mad/robopaint #208: Loading modes from a URL or folder

docprofsky, 2015-02-17:
It would be nice to add the ability to add a mode from a URL or folder. This way people can use modes not included in a release of RoboPaint and not have to run it from source.
Brackets has implemented this in a manner which seems suitable for RoboPaint. Brackets allows you to select an extension (or mode in the case of RoboPaint) from the a file or URL or from the Brackets Extension Registry. An implementation for RoboPaint can omit a mode registry.

Features required for external loading of modes in RoboPaint:
- Loading from file / URL
- Add a mode manager to add, remove, and disable modes

[image]

rogerfachini, 2015-02-17:
One thing we could create is a website/database of modes that users can upload to. Loading a mode via URL should be fairly easy to implement.

docprofsky, 2015-02-17:
It should be. All that would be needed is to get the file from the URL an unzip it.

rogerfachini, 2015-02-17:
It could also provide a way to find modes or ideas for modes to build as part of RP. The only downside would be a psudo-dependency on an internet connection for downloading new modes. Once the initial download of a mode occurs, an internet connection would not be needed

docprofsky, 2015-02-17:
One problem I see with this is management of additional Node packages required for certain modes.

rogerfachini, 2015-02-17:
Does JS or node have an easy way of unzipping files? It might be easier to have it  download an initial config JSON file, which would include URL links for the other resources

rogerfachini, 2015-02-17:
The node packages could pose a problem, can you install a node package from a command in js?

rogerfachini, 2015-02-17:
Is loading npm packages from source possible?

rogerfachini, 2015-02-17:
It appears so, for future reference: http://stackoverflow.com/questions/15957529/can-i-install-a-npm-package-from-javascript-running-in-node-js

oskay, 2015-02-17:
I'm not sure how much work it would be to build an interface like this, but I'm guessing "a lot."
There are also some security implications here that should perhaps be considered.

rogerfachini, 2015-02-17:
Security is another good point, we might want to have a flag associated with user-submitted modes on the database that gives a user a warning before installing a non-official mode. 

Most of this idea is very out of scope at the moment, but it is fun to at least discuss it.

techninja, 2015-02-17:
There are **_definitely_** security implications. NW apps have far more direct access to system resources, and really anything the application system user has access to (all hard drives/mounted locations, etc).

The following is my **secure & trusted** design for how a system like this might work:
* 1st party modes come bundled with the application and remain in the codebase
* 3rd party modes exist as their own repositories externally, with a similar structure to node modules (as they currently do).
* A mode is marked as "trusted" by the RoboPaint application, by first being reviewed by us, then added to a specific JSON file committed to the RoboPaint repo (which only trusted users have access to push to).
  * The JSON file contains download URIs, descriptions, preview images, and all other information required to display a preview of the mode.
  * This file can then be read locally by RoboPaint, or loaded remotely to keep updated as new trusted modes are added between releases.
  * To continue trust with each mode update, only the version specified by commit hash pulled from Github is given to users to install. This cannot be back-tracked or edited in any way to inject malware.
* **Dependencies:** As part of the earlier discussion, if a mode uses node_modules, how would RoboPaint install them without a local install of npm? Answer: It shouldn't. [Travis CI](https://travis-ci.org/recent) can install NPM and run `npm install` then we could push that "built" output to yet another repository (Github repos: They're free versioned file storage!). All of this can be automated and has likely already been written as a workflow somewhere else.

**Upshots:**
* As long as we control the master files and check each release, we (our users) remain clean and safe.
* All of this process is completely manageable via freely available storage and services, requiring no new external server, authentication, or cost to us.
* It's made up of standard processes! If a 3rd party wants us to review to get a mode in, they just submit a PR as their application

**Downsides:**
* Users would have to wait for our security review for every single mode/mode update. Considering the size of our community right now, this is likely one of those "good problems to have", likely not to negatively effect that many users until sometime in the future when we both have a large active user base and thriving mode community.
* No direct URL pasting. I figure if people want to try something untested/insecure, they can just screw with the files/folders themselves. RP itself would continue to work in the exact same way it currently does loading modes from the `modes/` folder, approved or not.

This isn't in the current or _next_ milestone, but maybe after getting some specific feedback relating to extra functionality that could be brought in from actual WCB/RoboPaint users, we can gauge how important something like this might be for them.

techninja, 2015-02-17:
#### Actual development required to implement the above plan (how hard is it to do?):
**High Level**
* Plan out the structure and content within the master modes JSON file. How does translation fit into it?
* Documentation, documentation, _documentation!_ For the system to be usable by developers, it must be clear how it all works and fits together, not to mention modes likely need a better API and to be more stable. 
* Find people actually willing to participate as active 3rd parties for a mode.

**Code/Development:**
* Write a browser, selector, and loader for the JSON file. The design would likely be based off Brackets or other "App Store" style interfaces. Comments, ratings and the like are even possible in these circumstances, but could come in _waaaay_ later.
  * Assuming we could agree on how it looks/works, I don't consider this all that scary. I've built 100+ similar "applications" in JS.
* Write the automated Travis CI grunt scripts to build out approved modes for download.
  * I've attempted similar things and people do these kinds of things all the time, so this isn't too tricky.
* Cleanup/standardization of the modes APIs.
  * Again, the only scary part here is working out what things we can/should make easier for mode developers.

docprofsky, 2015-02-17:
I just want to be able to load a mode from a file or URL. I do not like the inability to load directly from a URL. I think having an "App Store" is overkill for the current state of RoboPaint.

techninja, 2015-02-17:
I think loading from a URL is overkill to be part of the project itself. You could likely write a commandline script in an afternoon that would do it easily. Truth is there are no other modes outside the codebase yet.

techninja, 2015-09-23:
Ok, so... This has somewhat been implemented. Modes are now loaded from the node_modules folder at the root of of RP, basically meaning that modes are now their own Repositories, and can be managed by NPM for free! Setup your account with them, setup your repo with your mode in the root, and run `npm publish`. Make sure the mode is in the format `robopaint-mode-[modename]` and our global namespace will be preserved.

See all of mine here: https://www.npmjs.com/~techninja

When that's done, you can install a mode into RoboPaint by just running `npm install robopaint-mode-[modename]`. _Amazering._

closed 2015-09-23 by techninja