evil-mad/robopaint #212: SVG file creation not properly initialized. (create mode)

oskay, 2015-03-17:
Drawings created in Create mode save poorly (with malformed SVG), unless you first select New Document or open an existing file.

To reproduce: 

(1) Launch robopaint. Make simple drawing with just a circle, try to save.
-> Result: Bad SVG file, with extra `<rect ...>` and ` </g>` before first `<g>'.

`<svg width="10.99986in" height="7.998293in" xmlns="http://www.w3.org/2000/svg">
   <rect fill="url(#gridpattern)" stroke-width="0" y="0" x="0" height="100%" width="100%"/>
 </g>
[...]
</svg>`

(2) Try to load that file. 
-> Result: Cannot load file (since it's a bad SVG).

(3) Make new document, or open existing document, and try to save
-> Result: Success; makes working SVG.

(4) Restart RoboPaint, File > New Document
- > Result: Saving works OK.

techninja, 2015-03-17:
Ouch, yea, that's real. Thanks method-draw! Sheesh. Will fix in 0.9.6 as listed.

techninja, 2015-05-06:
@rogerfachini has claimed this one, a PR is likely soon forthcoming.

techninja, 2015-09-23:
This issue was moved to evil-mad/robopaint-mode-edit#2

closed 2015-09-23 by techninja