evil-mad/robopaint #213: Add unified calibration wizard overlay

techninja, 2015-05-06:
All the modes use their own somewhat unintuitive buttons for pen/brush control that are intended to be used for debugging & calibration, and really every mode needs some kind of method of doing this if intends to paint/move the bot on its own. Doing demos on applications that draw and don't have a calibration routine/buttons means either fiddling with a moving pen, or playing around on the CNCServer localhost page. Almost manageable for dev use, but still pretty crappy from all sides. Not to mention the more re-use we get, the more useful & unified things can be.

I'm imagining something akin to the simple step by step interface window from #130, with clear images from the wiki showing what to do when for particular bots. We might not even need to run the unlock, just simply move to the "corner" of the page for WCB, or perhaps the center "park" for EggBot. 

I would likely not remove any buttons from the print mode, just add a new one for "Calibrate" below the others, and would have a new toolbar button to the left of the question mark. Neat thing is this would be a perfect time to ask the user what implement they plan on using, changing settings defaults for overshoot, etc based on brush length, pen type, etc. 

Will post/take suggestions of what the "wizard" steps might be.

docprofsky, 2015-05-21:
I think "Calibrate" is a suboptimal term for this and "Align Brush" is better.
We should implement this as a frame/HTML element that can just be put in any mode easily and they will all have the same look.

techninja, 2015-05-21:
If modes live outside of the main code (as intended) then I can't recommend that they have elements to exist or be reused within modes, but for the act to be ignored by the modes, and an overlay that exists in the same HTML space as the settings menu (outside of modes) would appear over the top. I think settings is overused and cluttered as it stands, so moving more useful globally useful things out into the toolbar space isn't such a bad idea. 

"Align Brush" doesn't take into account other implements or methods of operation. Perhaps "Align Implement"? The term "align" to me says "X/Y matching", not exactly the Z/height verification that we're actually doing. Maybe mentioning height here isn't such a bad idea.

docprofsky, 2015-05-22:
I agree that brush is less than great, but implement feels clunky.

techninja, 2016-07-30:
I did this, and I don't think people actually like it. I'm going to take votes in here from @docprofsky @oskay etc if this should be left in after Beta 2, or heavily modified. I'm very open to suggestions.