evil-mad/robopaint #218: Roll 0.9.6 bugfix only release

techninja, 2015-05-20:
_This issue is a placeholder for the work and discussion behind the next release and should act as a tracker for any release delays or important points about the build not already ticketed._

This release supersedes the previously planned 0.9.7 feature release, fixing more critical/important issues above feature additions. 0.9.7 was set drastically off track by numerous factors involving feature implementation complexities and personal issues, leading up loss of my main source of income early in April.

0.9.6 has been identified as the plan to fix any major blockers before the next feature release, though it's been hard to pin down everything that goes with it, the milestone should hold the basics of what's required before release, though not the timing.

Whatever I seem to set as the goal just slips away, though I knew April-May 2015 would be tricky, I had no idea I'd get thrown for yet another loop in the form of trying to find a job. Regardless, the last major time crunch is upon us with my leaving for a whirlwind book tour with @makersylvia across Australia for 2 weeks starting May 29th (returning June 15th or so). Of course if there's a power outlet available on the 18hr flight, I might actually be able to get some work done ;)

Will list the tickets of issue and their status later on, I just wanted to make sure this issue was created at a minimum.

techninja, 2015-08-01:
### Big Update:
Obviously all deadlines have been thrown out the window. Constant intrusions into daily life have destroyed my ability to keep this update on track. The good news is great strides have been made:
* I didn't intend it, but we've successfully moved the entire system off of nw.js. see #215 & #221 
  * It runs in a very different way, has a very different active group behind it, and better than everything else, it's faster. Big improvements will continue to be felt, including (eventually) automatic updates!
* A branch that speeds everything up when it comes to slow fills/tracing is being revived and should be out soon (it was waiting on Electron to finish), this will fix a number of bugs and really improve performance in general.
* Biggest priority now is to get 0.9.6 out with the fixes needed for clean working scratch support  #209. Not that I have a complete answer there, but we're working hard on it, not to worry.

I've begun work on a new Electron project, the [PancakeCreator](https://github.com/PancakeBot/PancakeCreator), and though it's in its infancy, it will be blossoming in the weeks to come and will surely bring new and amazing features/ways of doing things cleanly and quickly.

oskay, 2015-09-21:
Can you please build a copy so that I can test it with Snap support?

techninja, 2015-09-21:
Will do tonight. If you're testing other things as well, currently interesting/missing/wrong/spotty:
* Fill/stroke settings (They're setup internally with defaults, just haven't redone the settings dialog)
* Controlling machine path preview visibility in print mode. Not really sure what convention to use for that. Maybe some buttons. 
* All commands fully buffered by default before starting. No more weird pauses during initial brush wash/painting.
* Fill and stroke in general test wonderfully, but should be tested as much as possible.
* New CNCServer fix prevents wackyness with lag that might help some other things.
* Example rebuilt text mode is very decent, but I haven't done all the buttons yet.

oskay, 2015-09-21:
I'm headed off to MakerFaire NY on Wednesday, so I'm not sure how much time I'll have to test it this week, but I would appreciate the build, and I will look at each of those things.

techninja, 2015-09-21:
Awesome! For fun I'll include Spiral mode for playing with.

techninja, 2015-09-22:
Ok, Here it is: [RoboPaintMac_v0.9.6_TEST.dmg](http://robopaint.tn42.com/RoboPaintMac_v0.9.6_TEST.dmg)

This pulls all modes from actual separate repos! It's amazing, and awesome. Check out my new NPM namespaces: https://www.npmjs.com/~techninja
**Also:**
* [Spiral paint mode](https://www.npmjs.com/package/robopaint-mode-spiral) is quite fun. ANd yes, it's jittery. This is to do with all the tiny "waits" strung into the system for pen up/down. It's correct that we should wait, but I suspect that for this use case everything would act a bit smoother without them. I just worry it won't be where we expect it. Also, this mode doesn't have a real icon yet, looks like another print icon.
* Text mode remembers all its settings, but again, doesn't have all its buttons. It will happily print or save to SVG your text though. Also its sidebar is too long and gets covered up by the new status popout.

There's a number of things that are wonky, but known:
* Closing doesn't always seem to close (I blame some stupid logic I have yet to fix)
* There's no way to hide the print mode previews once they've been made.
* Manual mode is all kinds of broken
* Remote paint mode isn't even nearly working (it's on the list).

And more, I'm sure. Also I don't think this is 0.9.6. This is bound to be a major revision, possibly even 2.0 if we decide it because it's so massively different, and even likely to automatically close a number of tickets in the issue queue right now.

oskay, 2015-09-23:
You sure closed a heck of a lot of tickets-- it took me a long time just to _read_ through them!

I'll prioritize testing here as soon as we're back from Maker Faire. :)

techninja, 2016-01-13:
Closing this for the sake of 2.0 beta and future releases.

closed 2016-01-13 by techninja