evil-mad/robopaint #222: Windows Electron install  & EBB driver

techninja, 2015-08-11:
One thing different in the new Electron Squirrel install, is the _missing_ ability to add other things to install. Everything beyond getting the program installed, available to users and running, is _**extra**_ and needs to be added to the application logic itself, so it begs the question:

#### How should we handle driver install?
**Options:**
 * Include the driver executable with the application files, attempt to execute (with admin rights??) when prompted
 * Prompt the user to click a link to download the driver installer right there (or do it for them and send them to the file location in Windows explorer)
* Ignore all of this and put a link near the download page for the driver :P

**Open questions:**
 * Can we detect if the driver is installed?
 * Failing that, should we suggest the user install the driver blindly if the bot isn't detected? Doesn't hurt to reinstall.

EmbeddedMan, 2015-08-11:
I don't know of an easy way to detect if the driver is installed. There may
be a way to query a registry entry or something, but it would likely
involve considerable understanding of how each Windows version stores
information about installed drivers - and I don't know how to find such
information at this point.

We do have the source for the USBDriverInstaller.exe application, and so -
with some work - could possibly extract the calls being made to the Windows
driver installer APIs and make those calls ourselves from the code that's
run during the new install process - if that's even possible - or when the
application first runs. That seems like a lot of work for little gain
though.

It feels to me like simply downloading the USBDriverInstaller.exe for the
user and then telling them to run it would be a good way to work around
this limitation in the Squirrel installer.

It is true that it doesn't hurt to reinstall.

But one problem with this plan - if a user wants to install this software
on a non-internet connected PC, they're out of luck if they need the driver
installed. This seems to be a nasty limitation.

*Brian

oskay, 2015-08-11:
I like the idea of including the driver executable with the application files.  Running the installer could be done as part of the "first run" setup, perhaps?

techninja, 2015-08-11:
I too was a bit worried about portability to non-internet connected machines. Increasingly rare as that may be, it still happens. Packing the installer in the source means that there's only one file to get (at the "time of getting"). I think what we're missing here now in question, is the exact logic behind inviting the user to install the driver.

How about this first install confirmation box before bot detection:
**_"Welcome to RoboPaint! Before you can start using your WaterColorBot on this computer, we need to install the USB driver support software. Click OK to continue with install, or click Cancel if you're sure you already have it, or just want to try out the software."_**

Eh?

EmbeddedMan, 2015-08-11:
And then when they click "OK", you can run the installer? That seems like a
great solution.

*Brian

oskay, 2015-08-11:
A possible issue is that there should be a straightforward way to do it later, if you need to not install it at that time.

A possible approach would be:

"Welcome to RoboPaint! Before you can start using your WaterColorBot on this computer, we need to install the USB driver support software. Click OK to continue with install."  [OK][Do not install driver]

If [OK]: then run installer.

If [Do not install driver]:  Dialog box: 
"You can install the drivers later by going to: Settings > Basic Setup > Install USB Driver." [OK]

Is it straightforward to keep track of whether or not the driver installer has been launched (from within RP)?  If so, then future "can't find the robot" dialogs could also indicate: You may not have the USB driver installed.

techninja, 2015-08-11:
Yup, all quite doable. Our persistent storage allows for setting of arbitrary anythings, so that is an easy win. Sounds a little complex, but not technically (in as much as properly identifying if they already had the driver could have been).

Thank you @oskay & @EmbeddedMan, the remaining work on this issue are for me to actually implement the agreed on path forward :)