evil-mad/robopaint #228: RoboPaint NG Release feature implementation checklist

techninja, 2015-09-27:
Since the creation of the PaperJS branch, I've [pushed over 170 commits](https://github.com/evil-mad/robopaint/compare/paperjs) to basically do a **heavy** refactor of the entire RP system to fit better with the way that Paper.JS and Electron play together. I'm creating this issue to whittle down and make it clear what's left before this can be fully released. These are mostly in order of priority or how I intend to write them, top to bottom.

**Drawing**
 * [X] Implement Paper NG stroke & fill in working modes with new settings (#226)
 * [ ] Implement Paper NG stroke & fill settings pages with preview (#226)
 * [ ] Implement some kind of _basic_ media creation utility for custom colors and pen color groups (with defaults)
 * [ ] Make multicolor pen (wait/resume) tool changes for color work (with calibration), likely requires unified messaging below...

**Mode Support/Mode API**
 * [X] Unified cancel, form settings management, and cncserver control
 * [ ] Clean, nice looking unified modal messaging. [Electron's MessageBox](https://github.com/atom/electron/blob/master/docs/api/dialog.md) _might_ be good enough. It is native.
 * [ ] Semi-automatic qTip for unified hover titles (support base modes at least)
 * [ ] Support for mode color SVG/png icons, & "teaser" image
 * [ ] Persistent API extensions via package.json (for #227, etc)
 * [X] Convert Print & Edit mode for NG
 * [ ] Convert Example mode for NG (~90% complete)
 * [ ] Convert Manual mode for NG (~50% complete)
 * [ ] Convert Remote Print for NG Mode 

**"Beyond" features:** These have been on the agenda for years, but have never become anything too real. I feel as if the time is about right, but it will take more time, and it does mean less effort given to ensuring basic features are as absolutely polished as they can be.
 * [ ] Merge in basic multi-bot support. This has stagnated in a branch confused by design, and maybe it just needs a kick in the pants and some time to play around with the implementation a bit.
... More to come.

forresto, 2015-09-28:
Excited for multicolor pen (wait/resume) tool changes. One suggestion related to my not seeing the overshoot option (#229): switching to pen mode could also turn overshoot off.

techninja, 2015-10-13:
Bleh. "example"/Text mode is feature complete now, and really quite useful and fun, though the UI design kinda sucks, but it's not like that's any kind of surprise. How the heck do you fit in that many sliders into one place? Sheesh. 

[image]

Questions/comments/PRs for this mode should go here: https://github.com/techninja/robopaint-mode-example