evil-mad/robopaint #229: overshooting path

forresto, 2015-09-27:
Seems to overshoot every path by ~1cm:

[image]

Is this a Robopaint or firmware bug?

techninja, 2015-09-27:
Hello there @forresto! The overshoot in RoboPaint is a setting made to help painting with bent brushes, but is terrible for pens. Go to the gears (Settings) page, and find the "overshoot" option, and you should be able to set it to 0. If you're having trouble I'd be happy to provide you with screenshots. Good luck!

forresto, 2015-09-28:
Right! Feature not a bug. Thanks.

techninja, 2015-09-28:
Yep! Next version will turn that off by default for pen/pencil mode. Thanks for the mode suggestion by the way. Feel free to suggest or bug us for anything you might need, we're only as active as our community!

closed 2015-09-28 by techninja