evil-mad/robopaint #230: z freaking out?

forresto, 2015-09-29:
Not sure if this is a bug or setting, but I coded these to be 3 continuous paths each, and it seems like it is randomly lifting the pen. https://youtu.be/1gaZM0wy450

(I dig it aesthetically, but...)

techninja, 2015-09-29:
Aha, yes. Love the design, very fun! What you're seeing is the pen trying to avoid going on top of other paths, hence the lifts. This is on by default on works great and as intended for painting projects, but is pretty silly for pen drawings. Unfortunately in an effort to keep print mode as simple as possible, we didn't add a checkbox for this. Again, in the next version, this will be highly configurable but for right now, you need to go to Settings > Advanced > Modes: Manual, activate that, then go to manual mode, and uncheck "Check stroke visibility". This will completely turn off any special occlusion checking (with the exception of fills), and should quiet down your drawing quite a bit.

_**Pro Tip:**_ When using pens, set your "brush height" to only just as high as you need it to be clear of the paper, and your drawing will be both quieter and faster as it doesn't have to wait as long for the shorter pen movement.

closed 2015-09-29 by techninja

forresto, 2015-09-29:
Interesting, so it tries to not trace the same path.

Suggestion: collect these pen protips in the Readme or on http://watercolorbot.com/ .

oskay, 2015-09-29:
The best place for these kinds of things might be here: 
http://wiki.evilmadscientist.com/RoboPaint#Troubleshooting

techninja, 2015-09-29:
Yup, wiki is best! But my goodness, that wiki needs some updating love. I have a feeling after we get this next big push out, at least a week should be spent getting all that wiki documentation and screenshots updated. There's lots of good stuff on there, but some parts relating to old software versions just need a bit of polish and screenshot updates.