evil-mad/robopaint #231: Lots of mode bubbles: Looking for a UI design answer:

techninja, 2015-09-29:
One of @MakerSylvia's original designs in RoboPaint was to have each mode in a bubble and in the toolbar. At _best_, the current toolbar & home screen was designed for maybe about 4 modes.

With the latest work in the new branch, we've got the ability to turn on and run at least 6 out of the gate! And @docprofsky is already working on one, and there's lots planned that could be made relatively easily with the powerful API I've built. So the question becomes: How do we allow for it? The UI may "physically" scale well, but the _**concept**_ falls apart after 5 or more modes.

[image]

**Possible toolbar options:**
* Overflow modes into some kind of dropdown arrow overflow menu. Selected mode icon is shown to the right next to the dropdown if inside menu.
  * Would likely need the ability to re-order modes to customize this.
* Toolbar section for mode selection (all the mode icons) simply becomes one big pretty dropdown selection (same width as 4, maybe after it gets past 4?), comes with name of mode, description, icon.
  * This probably falls apart past 10-15, but I guess it could be scrollable.

**Possible Home screen options:**
* Let the bubbles go to two rows. Only solves till ~8 modes. Could then go to three lines, but bubbles may still overflow. We'd likely need to have them shrink in size depending on available screen area.
* Change the bubbles to be a picture of what they output (already planned) but the bubble size and position on the home screen is based on mode usage stats, and positioned around based on a calculated force within the allowed screen area ([a bit like this](http://dn.ht/springy2.html)), with the ability to drag them around. Kind of more fun than useful, but that's kind of the point.

oskay, 2015-09-29:
Ah, like so!
[image]

techninja, 2015-09-29:
Hey, I'm not crazy! Yea, something like that... ish. I'm still finishing formatting the settings window and the new fill styles, but once that's done I'll likely try one of these out and post pictures here for audience approval.

oskay, 2015-09-29:
It might be helpful to give a couple of the standard mode higher status for the time being-- locked in place above the rest, for example.

techninja, 2015-09-29:
Completely agree. Curious, what might you choose as the "teaser image" to go in the bubbles for Edit and print mode? Screenshots could work. Or perhaps an actual photo of a good Mona Lisa painting for print, and for edit, a selected path on the Mona Lisa SVG?

oskay, 2015-09-29:
I'm a bit puzzled by the question of exactly what these modes do, and how they work. It doesn't quite gel for me yet. 

Create: OK-- get it, create artwork
Print: OK-- get it, print that artwork that has been created.

Spiral: ? Is this creating a spiral? If so, shouldn't it be under "create?" Or is it to print a spiral, in which case it should go under "Print?" Or is it something else entirely? 
Text: Same question

Manual: Manual what? Manual create? Manual print? (And if it's one of those, why isn't it a sub-mode within one of them?) Manual something else?

oskay, 2015-09-29:
Should also say: I expect that some of these are subtopics within either "create" or "print" -- maybe we should handle those that way.

techninja, 2015-09-29:
Well... Spiral mode is fully encompassed to both _create_ the art, and then _print_ it (there's no export as it contains Z information not obviously storable in SVG). It's technically an option for any mode to create art in SVG format, that can then be opened in print mode... but really the mode api has made it crazy simple to take the abstracted stuff that makes print mode work and use it, with per mode customizations, so the separation isn't clear.

Perhaps the problem with mode here is the "one word" description. If we could define what each one of these modes does in maybe.. 3-6 words, and then show that text on hover, but tease with a picture defining what its output is, we would improve both understanding, and likely translation (as single words usually don't translate directly anyways without context).

oskay, 2015-09-29:
>it contains Z information not obviously storable in SVG

It's easy to make custom SVG elements-- we use this within Inkscape to store settings and the last known pen ("turtle") position.

>Perhaps the problem with mode here is the "one word" description.

Yes. Maybe "Spiral Print" or something like that. And pictures would also be a good approach.

techninja, 2015-10-15:
Ok, so I talked with @makersylvia a bunch and I think we have a general consensus: "The bubbles are fun!" and, "Lets steal from the Netflix Webapp."

Here's a current draft of the bubble business and new icons (not final): 
http://robopaint.tn42.com/testbubble/

**Notes:**
 * Bubbles don't move if being hovered over
 * Bubbles make way and shove any non-fixed bubbles around based on a negative charge relative to their radius. Currently the only fixed bubble is the logo, but that's can be moved too.
 * We can have as many bubbles as we like here, though they'll likely have to change size depending on available window area.
 * Bubbles can basically be any size, but I've made the two main ones edit and print on either side of the logo a bit bigger.

**Full Intended Functionality:**
 * On first load, all loaded modes show with their bubble icons (same as the icons in the toolbar), and move to fit the space given.
 * Resting/moved positions will be saved for next load, so users can place them where they like, though as new ones are added this will be adjusted.
 * On full click (not drag), RP will simply switch directly to the mode
 * On "long hover", like in the Netflix app, a mode circle will expand with the "short info" text and 2-3 word mode name over the icon with a faded transparent black gradient, with two action buttons:
  * First, A "More info" button (A question mark? Three dots? It's a bent line arrow in Netflix), that will expand in some kind of larger space with the full mode description, and the preview image(s) that will come with the mode to show what it does. (This stuff will be very useful to have with a mode browsing system too!)
  * Second a "Go" button (perhaps a right arrow? or [something else](https://thenounproject.com/search/?q=open)). Something to get past the info and just launch the mode at this point. In Netflix this is the "Play" button, usually available on top of the screenshots/pictures at all times.

Feel free and respond with any gut reactions on the new icons, or the proposed interaction. I'll likely not touch this for a bit as I have to get the final PancakePainter release done, and there's a few more important things on the list above this.

oskay, 2015-10-15:
The initial "unprovoked" movement of the icons seems like an unnecessary distraction to me. It's nice that they can be rearranged... but since one does not *need* to move them, it feels like a bit of unnecessary friction that you have to think about whether one needs to move them. 

It would be nice if the mode name were *immediately* visible when you hover over one of the mode circles. Perhaps that label could go on a reserved bar of space at the bottom of the screen?

techninja, 2015-10-15:
The unprovoked movement at the start is a consequence of the initial positions not jiving with the algorithm. If I matched it, they'd be still at the start.

The ability to move them via click/drag seems neat, but yes, as you said, not a requirement. I'd suggest making it a right/middle click and drag for power users to make their home screens nicer, but not a standard/required interaction.

If we sacrificed a little room inside each bubble and stuck the "2-3 word mode name" at the bottom, I think we might be better off that sacrificing a whole bottom bar (notice modern browsers all got rid of their bottom bars over the last 6 years or so).

Here's both  mocked up:

[image]

[image]
I _think_ I prefer this one, but _oh goodness_, there's a big limit on how much you can fit in the bubble, and it's unlikely translation will be kind to our terse English here.

techninja, 2015-10-28:
[image]

### Heavy update: http://robopaint.tn42.com/testbubble/
* The supposedly required "jump" at the start I can apparently skip (so I have) by forcing the calculation till alpha gets below 0.01. Works great.
* The "really hard thing" with fitting/wrapping text has been solved by [D3Plus!](http://d3plus.org/). We're using the "text fitting" algorithm, which means no matter what text is given (in any language), it will always wrap and scale to fit. I suppose there's no guarantee of size, but given the set constraints of size, the text in the circle and detail will always fit.
* Logo is now centered
* @makersylvia Has designed this so the icons have the preview images behind them, but hover hides them and shows a lower opacity logo with the text over.
* This hover position (in the next iteration) will have a button where you can go directly into the mode.
* When clicked, you get a centered info "cloud" with the logo, a better view of the preview content, and detail text.
* SVG filters are crazy.

[image]
^^Where I have my mouse cursor here is where I plan to have the "go" icon, though we may want to make it more obvious on how to get to the mode/ get out of this"window"

techninja, 2015-10-29:
@oskay  "Go" Icon has been added to the TestBubble URI above, with hover effect. I think this is pretty much done. I'm looking to you for approval on the wording for hover and detail text. I know some of this is fudgey, but I'm looking for clarity and ease of translation.

oskay, 2015-10-30:
I think that this is basically "good enough" although I might prefer to have the names a little more visible, to help first-time users along. One approach would be to add some more intro material-- maybe tips on startup, or tips the first time that you open it.

Let's move forward for now, and come back to it later.

techninja, 2015-10-30:
That's what I figured. Iteration requires making steps forwards, and that's what we've got. Not the answer, but the start of one.

As far as helping users out, this seems darn awesome and can be implemented easily: [Intro.js](https://usablica.github.io/intro.js/)

closed 2015-10-30 by techninja

oskay, 2015-10-30:
Yes, that sounds like a reasonable approach.