evil-mad/robopaint #236: Errors with Hershey text from Inkscape

forresto, 2015-10-12:
1. Use Inkscape Render > Hershey Text
2. Save SVG
3. Open in RoboPaint ... the text isn't in the position saved, and dragging doesn't work as expected.
4. Switch to Print, get this error:

> We're sorry, but it looks like there are some incompaitble elements in your SVG that RoboPaint couldn't automatically fix. 
We suggest you open the SVG in Inkscape or another SVG editor, simplify the document or just resave it. 
You can continue and try to print this, but you may have issues. Error given below:
Failed to execute 'getItem' on 'SVGPathSegList': The index provided (0) is greater than or equal to the maximum bound (0).
Is it OK to leave/reload this page?

I tried manipulating it in Inkscape first (grouping, ungrouping, simplifying path), but it doesn't seem to help.

techninja, 2015-10-13:
Yep. The real problem here is with how method-draw (the Edit mode app) deals with incoming SVGs. It doesn't position them the same (Inkscape, Illustrator all have differing ideas about framing, not sure why), and then it just dies trying to manipulate or import simple things.

That's not to say I haven't done it, it's just annoying and isn't reliable.

**Some good news on the font front:** I was so annoyed by this, I made a text mode (enabled in the advanced settings tab) that lets you render Hershey text directly in a mode and output to SVG which is less likely to be yelled at by method-draw. Enable "Example text mode" and try it out. Important to note printing text, you really don't want to check for occlusion, so print it in manual mode with that setting unchecked, at least until the next version comes out with vastly improved text rendering mode and direct printing.

techninja, 2016-07-30:
I'm going to assume that being given the ability to create and export SVG text from within a mode in RP is enough to move on with this. If not this can be re-opened.

closed 2016-07-30 by techninja