evil-mad/robopaint #237: 2.0.0-Beta1 release checklist

techninja, 2015-11-01:
> *Making things... making things...* Wait, shouldn't we release at some point? ...maybe, probably... _**Yes!**_

After nearly a year of making great strides with the latest experimental version, before we get ahead of ourselves with wacky new features (#234, #228). we need to get something out there. Here's the bare minimum checklist (mostly for me) to get out a Beta1 release.
 * [ ] Verify wording for modes & update (#231)
   * What's displayed where should be finalized. We're probably going to drop the "word" key for modes as unhelpful, and keep the 1 word machine name, a 2-4 word "name", a 2-5 word "descriptive title" (for the mode bubbles), then a "detail text" ~1 paragraph or so describing what the mode is/does.
 * [ ] Finish Basic Dropdown toolbar for mode switching, OR allow the logo to drop away gracefully with 5 modes on (that's really the main problem limit as there will be no ability to add modes for regular users for a while).
 * [ ] **TESTING:** Need to print all current example images with three different fill settings on Windows/Mac on default speed setting.

... I think that's it? I can likely do the release _before_ all the testing is done, but I suspect there's a couple edge case "stupid level" errors still lurking in there as I test directly with the bot and brush so rarely.

techninja, 2015-11-03:
Initial testing has gone well, but I've run into a snag on canvas paths that needs to be worked through. Good news is the various fill/stroke settings and styles add a lot of great style adjustment and seem to work pretty darn well!

techninja, 2015-11-03:
Testing done (enough), I found a _**VERY**_ longstanding bug in CNC Server where the draw speed was ignored, and squashed it flat! Also extra paths were being added to fills outside of the draw canvas, and that's been removed, so everything paints beautifully and incredibly smoothly. Tests with previous killers have also gone quite well.

techninja, 2015-11-04:
One last tidbit remains.... and it seems zeroing is effectively broken. Well, at least in any clear cut use case involving streamlined buffers, as it requires at its time of calling that no other items be in the buffer. Once I get this worked out and upstreamed, we'll be ready to go.

techninja, 2015-11-05:
Done. Building and writing release notes is now the name of the game!

oskay, 2015-11-05:
w00t!

techninja, 2015-11-05:
@oskay FWIW, Mac 2.0.0-Beta.1 is in the draft release right now. I'm not sure how you're going to sign it besides acting directly within the DMG (it is a drive image). We may need to modify the grunt mac build process to add in signing... though the builder has support to do it, I haven't added anything. See if you can't figure something out while I get the release notes and Lin/Win versions up there.

techninja, 2015-11-06:
Linux is probably fine, but Windows is a priority, and our stopgap solution with the npm cull completely breaks the electron build. Sooo.... I'm currently battling the NPM 3 build dragon and hopefully I'll win. If anything, I can just release 2.0.0-Beta.1 as Mac only for the moment. Feh, and I was going to have this out today! :(

techninja, 2015-11-07:
I've officially fixed/added since the "half" released version:
 * Fully closing the app wasn't quite working.
 * Kinda, sorta #238, at least the meat of it (being unable to, by default, to get an image from edit mode to print mode), and a little error that prevented a clean start during edit mode without an SVG loaded.
 * Added mode names to the top of the clouds and improved the glow for readability, along with some "helpful" cursors originally intended for the home screen.

Still battling:
 * Windows NPM path length issues.
 * The final solution for #238

techninja, 2015-11-07:
**Windows:** As "stable" as node v5 may be, npm 3.3.6 is barfing all over a standard RoboPaint install. Master works, the paperjs branch does not. I've cut my losses and tried flatten again, and it appears to work, but now the app goes straight to black screen, I suspect path handling may be the culprit, but something wacky is going on. COuld also be Win 10 to blame, will test on Win7 as well.

Oh, and my Linux machine seems to have died. It's nearly 6 yrs old so.. I can't be too surprised :sob:

techninja, 2015-11-08:
**More Windowz Lulz:** Turns out nothing at all is wrong with the app at all. Everything runs great without the bot plugged in. As soon as it is, BAM, it dies. Likely right during auto connection start. Neat thing is this means I can debug it directly. Will report back.

techninja, 2015-11-08:
AHA! Found it, it was pretty serious.
https://github.com/voodootikigod/node-serialport/pull/623
That giant version leap to 2.0 in node-serial wasn't for nothing and I'm left to assume a lot changed and needs a bit of testing. Guinea pigs are we I suppose. Seems quite happy now. Will move on with the beta1 release.

techninja, 2015-11-09:
@oskay OK! Looks like all the issues have been worked, and only ONE new one added :P
* Windows builds with flatten have been made, and attached to the release for both 32 & 64 bit. No crashing!
* Latest Mac build with closing issues and other things mentioned above fixed also attached
* _**Apparently**_ Serialport 2.0.3 breaks only Linux with some somewhat awkward bug involving listing all the ttys, though 2.0.1 is fine... Feh. I'm omitting the Linux build for right now until I can figure out some kind of fix or patch. It seems simple, but it's not been fixed easily in my 10 minutes of messing with it.

Sign the Mac build if you can, and we can publish the beta.

techninja, 2015-11-10:
FWIW: A node-serialport maintainer called me on the above. Looks like a common bug so the wheel is squeaking a lot louder than just me :)
https://github.com/voodootikigod/node-serialport/issues/610

techninja, 2016-01-13:
@oskay Just rolled the latest Beta 1 attached to the original Beta release. Give that a test, a number of important show stopper bugs should be fixed... though I haven't been able to reproduce _the blood taker_, #250. 

If this release is good enough for more testing by the populace, sign it, re-upload and I'll launch.

oskay, 2016-01-13:
Testing looks OK so far. Now that we've made some progress, let's please roll #250 into this.

oskay, 2016-01-13:
I have not met all of the testing requirements, but things appear to be working on the surface, and my short list of showstoppers have all been taken care of. Go?

oskay, 2016-01-13:
Wait-- up to me, right? I'll see what I can do to figure out how to sign this DMG. ;)

oskay, 2016-01-14:
Signed and uploaded!

techninja, 2016-01-14:
## _WOO TEAMWORK!_
[image]

closed 2016-01-14 by techninja