evil-mad/robopaint #238: v2.0.0: Unable to transfer SVG directly from edit to print mode

oskay, 2015-11-06:
To reproduce: Open an SVG file in edit mode. Go to automatic image print mode. The SVG does not come along. (The previous SVG is still there.)

techninja, 2015-11-06:
I believe this is actually a bug with "empty" canvas. It errors out in some way, but works enough to show the art, but not save it to be transferred. Try opening an SVG from the home screen, it'll open in print mode, then go to edit, and it'll work.

If you can confirm this dies only with an empty canvas, this should be opened in https://github.com/evil-mad/robopaint-mode-edit

oskay, 2015-11-06:
No... I tried this two different ways. First with a blank canvas, and then reproduced it by trying again with two different SVGs. Also, there is no "dies" here-- just the wrong thing on the canvas.

Also, I don't see any way to open an SVG from the home screen.

techninja, 2015-11-06:
Quickload. Turn on "debug" mode in advanced settings and you should see a "Cannot read property 'indexOf' of undefined" error when opening edit mode with nothing already "opened". I can confirm the issue, don't quite know the cause.

oskay, 2015-11-06:
That sounds like an unrelated issue; I'm talking about a properly-loaded image in edit context not being available in the print context.

techninja, 2015-11-06:
I'm talking about the same bug actually. If edit mode starts without an image, it can't save the image. If it starts with one loaded from quickload, it'll be able to open one just fine and ensure it replaces the existing one.

oskay, 2015-11-06:
I was actually able to save the file (and verify that it was good by opening it in another session).

techninja, 2015-11-06:
Ok, the issue IS different. Has to do with the "save warning". Huh.

techninja, 2015-11-06:
OK! Two things: First issue is where the mode API is trying to hand the svg load an undefined. that's bad, and has now been fixed so `robopaint.svg.load()` will always return valid svg now :)

Second: It is currently incorrect, but the direct issue you speak of actually boils down to one of _**logic**_. The edit mode knows that the "save last image" is turned off, which means if you exit the mode without having saved your SVG out as a file, you'll lose any changes it has if you were to close RP from home or any other mode. There's two clear options:
 0. Provide a nicer dialog with buttons that clearly give options, EG: "Cancel", "Transfer image", "Save and continue" _(< A little opaque, but much clearer on user intent)_
 0. Just save the image to the mode transfer regardless of what a user wants to quit or not. _(< Easiest, though less clear on intentions. A user could click to leave a mode, then click cancel, but the image at that moment would be pushed to the transfer)_

techninja, 2015-11-06:
We can continue to discuss the best option here, but for the sake of brevity/action I've implemented a save on close only for both "open last" states. Rebuild coming.

oskay, 2015-11-06:
I interpret that "save the image to the mode transfer" means that it stores the SVG to a temporary storage location, which is then opened by the print mode. It seems like that would properly replicate the behavior of older RoboPaint. 

If we do *not* do that, then it becomes necessary to manually save to disk before printing...?

techninja, 2015-11-06:
Manually saving your SVG is not (and wouldn't be) required. Saving to the temporary location _is_ required, but wasn't happening before evil-mad/robopaint-mode-edit@ad6a7f8 above.

The logic behind the dialog was that _who knows_ what could happen if open last image isn't on. If you just created this SVG from scratch in the editor and left the mode, none of the other modes care for the SVG content, and RP itself will delete it on app restart.

We likely should on exit save to the temp area regardless, but make it clear to the user what consequences there are through a better dialog. A draft:
[image]

That got a bit wordy :P

oskay, 2015-11-06:
If the act of transferring to print mode and back is potentially nondestructive, then I would prefer to simply save to the temporary storage location, transfer it that way, and forgo any dialog box here.

My reasoning is that from the user perspective, they are not quitting a program, just switching between modes within RoboPaint. They would expect a dialog box asking whether to save or not before quitting the program, but not when going back and forth between print and edit contexts.  

The hitch is that certain modes -- like the spiral-photo mode -- do not expect to be handling SVGs, and we may _need_ to force the user to answer a dialog box about whether to save or not before leaving the contexts where SVGs can be passed back and forth.

techninja, 2015-11-06:
Currently, and a little different than before, no context is given to the mode about whether this is a _global_ app close event it's getting, or if the user is just switching to another mode. Thing is, this is functionally identical as far as the mode is concerned. it's being completely closed and unloaded from memory either way.

_**Perhaps**_.... What we're looking for here, is the logic for "open last image" to be reflected not in any given mode, but in RoboPaint itself. It knows if it has an image saved in temp storage, and it knows that with open last off it's going to delete it on app restart, so it can ask above anything else before app close and say something along the lines of: 
[image]

Kinda reminds me of the clipboard warnings on Illustrator or Photoshop.

oskay, 2015-11-06:
Yes, with careful wording, something like this could work well.
[image]

techninja, 2015-11-06:
The dialog in my last comment is a **global** one, meaning it would happen when closing the app from a mode, or from the home screen, and if you're in a mode not loading the current SVG (like text mode or spiral), you're not going to see what image happens to be in that temporary space, so you can't really talk about it as if they were looking at it.

Your dialog makes some sense for exiting Edit mode, though guaranteeing to a user than their
> "changes will be _lost_"

Is a little harsh and misleading if their image is still being transferred to another mode, even if it's lost forever when you close the app when not saved as a file.

One major benefit for the global dialog is that modes that create SVGs don't have to worry about bothering the user to save before moving to another mode. One bad thing here is that there's no guarantee the user remembers or can see what data is in the transfer space when being asked if they want to save it (this could of course be alleviated by some means of showing the user what it is before while prompting with the dialog...).

So which path should we choose: Global dialog, or per-mode dialog? Your phrasing in the last comment makes me feel like that's a "global" dialog for each mode, meaning we check to see if the SVG content changed while in that mode prompt the user when exiting it like the above. Thing is, without more logic about the file that's being saved, we might end up spamming users every single time they change modes and make tweaks.

oskay, 2015-11-07:
I meant that type of language should be used in any case where the SVG data would be lost.

oskay, 2016-01-11:
Has this been implemented somewhere? It looks like it is not in place, as of Beta 2.0.0.

techninja, 2016-01-11:
The dirtiest basic form of this is via 5d176e6, which simply ensures that the bug portion of this is done. I haven't caught up to where we left this issue, but from what I can recall, we didn't have a final simple answer on how this should be handled right now across all modes (or just the ones we manage/care about)

oskay, 2016-01-11:
Right now-- so far as I can see -- there is no way to actually print a file that you've just created is to save the file, and then to separately open it. It won't print unless you also open it (to clobber the copy in memory).  I thought that this was the original bug described in this thread, and it is marked closed.

techninja, 2016-01-11:
That's because it had been fixed at that point. Perhaps this is a regression. Will Check.

techninja, 2016-01-11:
Just tested:
 0. With "Open Last image" off by default, go to edit mode
 0. Open an SVG, do stuff to it.
 0. Click on print mode.
 0. Dialog comes up cryptically describing a possible future where you will lose your work if quitting from another mode.
 0. Cryptically we click OK to accept that possible future... and somehow our second cousin disappears entirely from this timeline :hourglass_flowing_sand:
 0. The image shows up in print mode :confused: 

The issue may be an unhappy SVG killing the export and not properly clobbering? Not sure.

oskay, 2016-01-11:
When I try this, I don't see any of the changes made in edit mode appear in the print mode-- just the file, as it was initially opened.

techninja, 2016-01-11:
Hmmmmmm.

techninja, 2016-01-11:
Just tried it with the lion.. worked fine. I'm suspecting this has to do with the file. OR it's a mac only issue? Will test more tonight. Reopening for the sake of repeatability on your side.

techninja, 2016-01-13:
Just confirmed this on Mac. Very annoying, and easily blocking release.

techninja, 2016-01-13:
Ok, now THAT's weird. Apparently I had saved the change back in November, but never published it! Whoops. Rebuilding and trying again.

techninja, 2016-01-13:
OK, fixed in latest rebuild. Admittedly the dialog absolutely sucks in its current state, but unless you have a clear stance on the above this will stay for the beta.

oskay, 2016-01-14:
Marking this closed; I will open a new issue towards fixing the dialog box.

closed 2016-01-14 by oskay