evil-mad/robopaint #240: Add option for insetting fills

oskay, 2015-11-10:
When drawing or painting, Hatch and spiral fills can often intersect with or exceed the boundaries of the original area to be filled. We compensate potentially for this by performing an "inset" operation on the area to be filled before filling it with a hatch.  

http://wiki.evilmadscientist.com/Creating_filled_regions#Using_Hatch_and_Inset_together

It would be nice to have an option to inset fills in RoboPaint, with an option in the settings to adjust the amount of inset.

techninja, 2015-11-10:
What range values do you want for the dropdown? I'm assuming we'd want negative inset as well.

oskay, 2015-11-10:
Not sure-- also not sure how easy or hard this is. This may be a little tricky to find a good set-- for pen/pencil you might want a much finer range of inset values than you want with a brush.  Perhaps powers of 2, for some units?  0, 1, 2, 4, 8, 16, 32, 64?

I don't immediately see a use case for "outset." Perhaps we should leave out negative values.

techninja, 2015-11-11:
Huh. I'm kind of flabbergasted, but not only is my modularity _**off the hook**_, it's damn extensible too!

[image]

Sure, I had already simplified the working JS Clipper code to do path offsets, but to be able to simply drop in a path preprocessor, setting, and brand new library without batting an eye and less than afew hours work to make something that feels like it completely changes the game?! This is crazy. This looks like how it's _always supposed to have looked_. **GEEZ.**

[image]

Up to 64 seems a very safe range, though I suspect you'll want to mess with it. The fills completely disappear for the ALL the objects in the preview at that point. Here's what the earth looks like with 64 unit inset:

[image]

I guess I'll finalize the text and push it out. While I'm at it I'll see if I can't get a bit closer to the Inkscape smooth fill shape.

oskay, 2015-11-11:
_Damn_ that looks nice!

EmbeddedMan, 2015-11-11:
Agreed! This interface is looking super crazy awesome.

techninja, 2015-11-16:
This is done. Will be in next build.

commit e3f8e5b7 (2015-11-11): Add Clipper Library and path fill inset support, fixes #240
closed 2015-11-16 by techninja