evil-mad/robopaint #242: Super Simple Pen support Proposal

techninja, 2015-11-18:
It seems I've gone off the deep end with my musings on an all encompassing system like #234, when really we've been wondering where simple & uncomplicated pen support has been for years.

Truth is, it's not far off, here's an attempt to describe the transition to a minimum viable solution:
* Add a new colorset type, `pen` (stuck in colorset config JSON)
* `pen` type sets display slightly differently, with a generic pen picture and custom color, only selectable when in pen/pencil mode.
* Tool change between colors is "pause/wait for change". Will require RP wide notification for which pen to put in, and shove to calibration before continuing. 
* Provide the following basic pen sets:
  * One black pen, default. Description will detail that if you only have one thing, it doesn't matter what it is, just use this.
  * Basic Color Pen Set: Same as crayola watercolor set, just with pens
  * Custom pen set. Need to come up with a very basic definition of what this is, but the bare minimum we need is a color picker and a pen name to go with it.

oskay, 2016-04-19:
This is an excellent idea. I'd be very happy to have this functionality present, even if we don't yet support custom pen sets.

techninja, 2016-11-06:
@oskay Finally got some time to get this silly thing out of the way.  This is what it looks like when a manualswap is triggered. the `__color__` would be replaced with whatever the translated RP name for the given tool index is (99% of the time is going to be a color).

[image]

Text is.. OK at best. along with the tool definition we can tell what this is, so I can have it say "...insert the blue pen/pencil" instead of awkward "..insert the blue color" :confused:, though I'll hold off till I hear your suggestion.

The use of "unlocked" is to remove the "unlock" button, so by default after we park, the motors will become unlocked for users who want to calibrate everything themselves.

techninja, 2016-11-09:
Apparently CNCServer API never had support for unlocking the motors while paused! Whoops. Turns out that's ab obvious requirement to do this at all. Sheesh.

techninja, 2016-11-09:
The PR for this work is in #287 

New screenshot: 
[image]

Full text for all the buttons/options:
```json
"manual": {
      "status": "Waiting for manual change...",
      "notice": "Please insert the __color__ __type__ to continue",
      "info": "Motors unlocked for calibration, use the toggle button find the correct draw height, ensure the pen is put back in park position, then click continue to resume drawing.",
      "resume": "Now drawing with __color__",
      "options": {
        "cancel": {
          "title": "Cancel Print",
          "info": "Cancel the current print job, close the mode and return to the home screen."
        },
        "continue": {
          "title": "Continue drawing with __color__",
          "info": "Once you've finished changing to the correct drawing implement, click to continue drawing."
        },
        "toggle": {
          "title": "Toggle height",
          "info": "Toggle the raising/lowering of the pen to calibrate its height."
        }
      }
    }
```

forresto, 2019-10-11:
FWIW, I did this manually by editing the SVG source, and switching pens 6 times. The alignment was a little unpredictable, but I find the imperfection charming:

[image]
– https://twitter.com/forresto/status/1182675515850874880