evil-mad/robopaint #243: Settings window: Mode toggle buttons are hard to actuate

oskay, 2016-01-04:
Tested on Mac. Some of the mode buttons are very challenging to press-- they seem only intermittently responsive.

techninja, 2016-01-04:
The way they're made likely has their two clickable elements moved around a bit. I'll verify and attempt a fix.

techninja, 2016-01-04:
Hmmmmmmm
[image]

Not quite seeing the problem. :confused: 

This was captured with "[LICECap](http://www.cockos.com/licecap/)" (_**terrible name**_, great app), with click animation on. See if you cant reproduce and record it and I'll see if i can tell what's going on.

oskay, 2016-01-04:
The problem is only in the _Modes_ panel.

oskay, 2016-01-04:
[image]

techninja, 2016-01-04:
Aha! That is actually on purpose! Core modes are not switchable. Notice they're somewhat faded to denote their disabled property. The idea being that core modes represent something that is expected and not directly toggle-able. I've been waffling on what to actually do for this, and tried it with having everything be adjustable, but you still kind of need them, so I just went with "disabled == core".. course it doesn't quite make sense for spiral... Eh. It's all a test.

oskay, 2016-01-04:
Ugh. You *do* realize that this means that they can't be turned on either, right?

[image]

oskay, 2016-01-04:
Related issue: The "Reset" button turns all of the modes off.  (This is probably not the intended effect.)

Once the modes are off, the core modes _cannot_ be enabled, except by quitting and restarting RoboPaint.

oskay, 2016-01-04:
Also: Is there any reason that the core modes must be left visible?  I can imagine situations where one might want to only allow (for example) the remote print mode.

techninja, 2016-01-04:
Aha!! THAT is the real bug there! Otherwise you wouldn't be able to disable them in the first place. What do you recommend (besides fixing the reset all mode disabled bug)?

techninja, 2016-01-04:
Yea, makes sense. We don't technically need the disabled locking. I'll just remove that and a user could choose to only have remote paint visible.. though clicking on a quick load image will still load the print mode

oskay, 2016-01-04:
Yes, disabling the modes is how I came across this. :)

Allowing you to disable any should be a great solution.... and auto-loading a mode like that when you ask for it sounds like a reasonable side effect.

techninja, 2016-01-06:
[image]

commit 61a5ee4f (2016-01-06): Set print & edit to be on by default on reset, fixes #243
closed 2016-01-06 by techninja