evil-mad/robopaint #244: Document API option to tell RoboPaint that it's ready to paint

oskay, 2016-01-04:
Following up on #224, this option allows the remote (or, programmatic) user of RoboPaint to allow new sessions to be opened without manual intervention.

techninja, 2016-01-04:
Took a second to look, forgot about the repository move! Does this cover us?
https://github.com/evil-mad/robopaint-mode-remote/blob/master/API.md#ready-state-override

oskay, 2016-01-04:
Yes, indeed! 

Is it OK if I replace the old API document ( https://github.com/evil-mad/robopaint/blob/master/resources/api/API.md ) with a note indicating where the current documentation can be found?

closed 2016-01-04 by oskay

techninja, 2016-01-04:
I don't think it exists in the [paperjs branch](https://github.com/evil-mad/robopaint/tree/paperjs). Maybe a new file? Or an addendum in the main readme would make sense.

I believe I'll be merging the paperjs branch to master as soon as we're good to launch the beta. IT IS, _the future_!

oskay, 2016-01-04:
My concern is that we *do* have this older document, that has out of date API information. That document should be removed, and if we remove it, it would be better to leave a "forwarding address" than to have a dead link.

techninja, 2016-01-04:
A valid concern. Though the document won't belong or exist there, I don't have a problem putting in a stub, I'm just curious where it might be linked that would require this?

oskay, 2016-01-04:
As of a couple of hours ago, it wasn't (so far as I know) clearly linked from anywhere-- an oversight. We do have some links to it here and there, and Google knows about it as well.

oskay, 2016-01-04:
Let's consider this _really_ closed now.