evil-mad/robopaint #249: Failure to connect on linux: Ubuntu 15.10

oskay, 2016-01-08:
Reported to us by @tobiasboyd, RoboPaint (2.0.0 beta) launches, but consistently threw up the No WCB detected - Simulation mode / Retry screen.

My testing on a slightly older system (Ubuntu 14.04) worked essentially perfectly; I have not been able to reproduce this myself.

techninja, 2016-01-09:
Here I go running the LTS just because I like the stability... and peeps just go flying off the handle and go next gen! I'm blaming node-serialport early on this one. :angry:.

oskay, 2016-01-09:
I am also not sure whether this is a "some computers" or a "some intrepid ubuntu users" issue. :P

techninja, 2016-01-13:
Oh hey, just stumbled on this!
https://github.com/voodootikigod/node-serialport/issues/621

Basically exactly the issue!! and it's still open. I'm going to submit the easy fix which is to just check for the existence of the string, not an exact match. Should work fine.

closed 2016-01-13 by techninja

reconbot, 2016-03-31:
@techninja I fixed it in master in serialport too, will make it into the next release

techninja, 2016-03-31:
Woot, Thank you sir! :smile: