evil-mad/robopaint #252: Improve dialog box when transferring from edit to print mode

oskay, 2016-01-14:
Following up on issue #238 which is now closed, it would be excellent to update the dialog box that you encounter upon leaving the edit mode. My suggested change is here: 
https://github.com/evil-mad/robopaint/issues/238#issuecomment-154493807

techninja, 2016-01-14:
I don't think we were quite done in that discussion. Rebooting it a bit, your suggestion:
[image]

This is the dialog that is intended to warn people that when quitting edit mode, their work might be lost.. though it won't technically be lost if transferring to another mode... of course if that other mode provides no save functionality, it's as good as gone if they forget to save before leaving.

The problem I have with the wording in the dialog you've suggested, is that it makes no mention of this transfer, and if they're only making a piece of art to be printed, clicking "Don't Save" is not inherently obvious that it's still allowing the transfer. Not to mention the point I brought up last time about "changes being lost", they will certainly be lost if not saved to a file before quitting, but not necessarily between modes.

I'm not saying I have a better/simpler plan... but with our current "wipe last image on quit" by default, we're asking for a little bit of complexity when it comes to explaining what the fate of edited images is to users.

oskay, 2016-01-14:
Looking at it again... perhaps you are correct. I wrote this dialog box back when it was the case that the artwork *was* lost when transferring modes.

oskay, 2016-01-16:
[image]

The current dialog box is shown above. It says, "Are you sure you want to quit? Your changes will not be saved before printing. Click OK to Quit."

This dialog *makes no sense whatsoever* in this context-- the user has not requested to print. Indeed, clicking "OK to Quit" is the only way that the user can get from opening a file to printing the file. It is deeply misleading.

As an interim fix, how about, "You have unsaved changes. Do you want to switch modes without saving?"    [Yes] [Cancel]?

techninja, 2016-01-18:
I am very aware how awful and stupid the text has become in the adjusted context, we just couldn't decide on the right way to talk about it what's going on and why... The other issue is that it uses the _**old**_ style confirm dialog that was limited to the standard "OK/Cancel" buttons.

The problem with talking about "switching modes" is (as discussed in the original issue), modes don't know _why_ they're being closed, just that they are being closed. Changing to another mode or closing the app with the `X` are the same. I can technically add the information about why/how a mode is being closed, and then we could write different text for each, but I'm not sure it's worth the trouble.

Can we come up with text that works for warning the user about _both_ quitting the app without saving, and the possibility that the temporarily stored SVG might be lost if another mode doesn't save it?

oskay, 2016-01-19:
Is a single dialog box used for quitting *every* mode, or is there a different dialog box for each mode?

oskay, 2016-01-19:
The purpose of the dialog box is to prevent unanticipated loss of data. 

- It is appropriate to give this warning when quitting directly from edit mode.
- It is *not* as necessary to give the warning when moving from edit to print mode, since we're not actually quitting... and data will not necessarily be lost.
- It is once again appropriate to give this warning when quitting from a different mode, that cannot save the file. 

If it's one dialog box for every mode switch + quitting any mode, that's a very different situation than if there's one dialog box for leaving edit mode...

techninja, 2016-01-19:
Currently (as discussed in the original issue), only edit mode presents a warning dialog when closing if not saved. There was discussion about moving this to be global for all modes, or making this a dialog specific to losing your "transfer data" SVG on global close, but we never fully decided.

I have a proposal, though it's a bit more complicated: We eschew the need for any dialogs, by keeping a full copy of the previous SVG every time a new one is saved. We can then populate a list of "previous drawings" (a bit like the quick load list, with previews), where a user can simply scroll through the list, click to load one, and possibly with an X to permanently delete one. This means we save everything at basically no cost to the user (less a little bit of HD space), and they don't have to be accosted by dialogs because they'll never lose anything.

Whatcha think?

oskay, 2016-01-20:
That sounds fine, although it sounds like _much_ more work than detecting whether you're quitting or switching modes.

An alternative suggestion: Change the dialog to read "Save changes before leaving edit mode?"

techninja, 2016-01-20:
I think it adds benefit without TOO much work, AND it solves the problem where other modes step on other mode content without warnings.

**Work Mockup:**
* Before mode save to data (through `mode.svg.save([data])`, if there's data in it, save that out as a file with the name as an MD5 hash of data, in a given folder in the RP user directory.
* "Recent work" dropdown lists all files in that directory (ordered by date updated), with previews (work already done for quick load)
* Click to load for a mode (already done for quickload)
* Click X to delete the file. Might have a "Are you sure?" Dialog here.
* All modes will already support loading/saving from this as it's built into the mode API.

**Foreseeable issues:**
* Might get too many in here eventually (thousands). I don't have a problem saving them all, but we may only want to show the top 100 scrollable. At least duplicate content files will always never have duplicates because of the hash filename.

oskay, 2016-01-21:
Keeping it to 10 or 20 documents is quite sane.

techninja, 2016-06-16:
Closing this as the PR is in #266

closed 2016-06-16 by techninja