evil-mad/robopaint #253: "Background" performance extremely poor

oskay, 2016-02-19:
I've been testing the Scratch/Snap interface, working on the Mac, with RoboPaint running.
Things seem to mostly be working when RoboPaint is the "foreground" app-- not running in the background. 

When a different application is in the foreground-- say, Chrome running Snap, or even just a text editor, painting continues, but with a delay of about 0.5 seconds between any two movements (line segment, lift/lower, etc) of the machine. For example, swishing the pen in the paint takes about 10 seconds.

In practice, this means that the Scratch/Snap interface only barely runs at all-- it feels malfunctional. If one switches back to RoboPaint in the foreground once the job is running, then everything works fine, but I don't think that this is a good solution.

docprofsky, 2016-02-19:
I have noticed this when using robopaint-mode-comic. It seems to have something to do with whether the GPU is rendering the RoboPaint window. When RoboPaint is not in focus, but still visible (partially or fully) it will be fine. When RoboPaint is no longer visible then I experience the same behavior as @oskay.

techninja, 2016-02-20:
Odd, and yet, not entirely surprising. I believe there's a flag for ensuring that we get higher performance whether the app is foreground or background, and can even add an OS "toolbar" mode, where the App is only an icon but still running in "foreground mode".

I'm going to mark this as high priority and might be addressing it a bit in my new cncserver performance branch coming up.

oskay, 2016-03-26:
It sounds like the performance branch is coming along nicely. Any guess when we might have a new build? :)

techninja, 2016-03-26:
It is. Though I've need to make time to prioritize it, and my day job has taken precedence. If you haven't seen, there's lots of chat on the [RP gitter channel](https://gitter.im/evil-mad/robopaint?at=56f04ad5745dcc317038bf3a), and we have some flavor of acceleration conversion actually tested working, though it assumes full deccel for every move, it's a great POC to show a further abstraction between "actions" and "command(s)" has made the CNCServer performance branch vastly superior.

It needs more testing and verification, not to mention I need to work out the three automatic modes of running a second process, right now it's being triggered and managed manually in both RP/electron and CNCServer console. Once that's done, I calm down the debug verbosity, and I write a few more basic non-scratch tests, we should be golden to get it into the next beta, likely within the next few weeks if I can devote some weekend time to it. It's like night and day to the previous version and will be well worth the wait and possible side effect issues.

oskay, 2016-03-26:
That all sounds great.
I would recommend using the acceleration only for _pen up_ moves, at least initially. (We have a branch of the Inkscape-based software (wcb-ink) that has that as well.)

techninja, 2016-03-26:
Got it. Obviously some things will need to be tweaked, @docprofsky Did a python->Js conversion and it definitely needs some adjustment.

oskay, 2016-03-26:
Yes; I had planned to suggest this (or possibly implement myself in js) once things were further along.

docprofsky, 2016-03-28:
@oskay these are the changes I made to add acceleration. I still need to handle short moves that from a curve.
https://github.com/docprofsky/cncserver/compare/performance...docprofsky:accelerated-performance

techninja, 2016-04-06:
@docprofsky What are your findings on the improvement from the performance branch? If you find it works we're one step closer to closing this :)

docprofsky, 2016-04-14:
I tested both native and non-native runners and there is no discernible difference between their performance. The bottleneck is in the mode sending the commands to CNCServer, and not CNCServer itself.

This flag fixes the problem.
`disable-background-timer-throttling`
http://peter.sh/experiments/chromium-command-line-switches/#disable-background-timer-throttling

commit 5487d53d (2016-04-14): Set commandline op for disable backgrounding
Fixes #253
closed 2016-06-13 by techninja