evil-mad/robopaint #255: Serendipity Seed : Art Creation and Print Production

RI0, 2016-03-04:
We had an interesting discussion the other night about the WaterColor Bot
and if it was able to create Art or if it was just a printer. We talked about  
the differences between creating art and creating prints. Although this could
be a bit contentious, one of the ideas about creating certain kinds of art was 
related the artist's or the machine's ability to reproduce it. 

Could we look at adding a Serendipity Button in the Print Mode 
that would render the current print absolutely unique? 

The button could open a window to generate the Serendipity Seed 
based on a set of variables such as :    
Time of Day, 
Current Cloud Cover, 
Current Temperature, 
or some other system based or human entered variable related to the
artist's mood. 

Some of the variables could come from an online service such as Wolfram Alpha   
https://www.wolframalpha.com/input/?i=weather+in+san+francisco

Serendipity could play out by adding some error to the brush height, the overshoot, 
the speed, precision, etc. in a manner that would make it very difficult to produce 
two prints that are exactly the same... and some might actually be better than the 
vector drawing source.

oskay, 2016-03-04:
That's a wonderful idea. We've noticed that -- at least when using actual water color paints -- prints often seem to be unique in their own way. The final painting depends on the level of paint and water, the types of paint used, and so forth. Adding some intentional variation (a "mood", if you will) sounds like fun approach as well.

However, it might also be worth considering painting modes that are entirely generative-- producing new artwork altogether.

RI0, 2016-03-04:
I agree that the prints are often nuanced for a variety of reasons...
usually string tension or what seems to be random re-inking with our bot. 

A full generative mode would be great if we can do it.
 
It may be easier to start with some extra +/- error based 
on some sort of unique starting place. It could be an audio file, 
a YouTube video, or a piece of data from one of the various 
open data programmes across the world :
https://data.cityofnewyork.us/Transportation/For-Hire-Vehicles-FHV-Active-Drivers/xjfq-wh2d 

It might be fun to have a Serendipity Setting in place before the 
Robot Art competition. 
http://robotart.org

PS - If we do look at a full generative mode, it may be worth looking at 
cellular automata on Scratch
https://scratch.mit.edu/projects/44786142/
and the Game of life on Scratch 
https://scratch.mit.edu/projects/45779124/

For reference 
http://www.wolframscience.com/nksonline/page-171 
http://www.stephenwolfram.com/publications/academic/cellular-automaton-properties.pdf

oskay, 2016-03-04:
This is off-topic, but string tension should not contribute to print variability (if it does, please contact technical support, or take out the slack in the cords yourself). Also, re-inking should be quite deterministic -- try running the same print twice, to see if it does re-ink at the same place. (You can just watch it with the brush out, to verify this.)  If there is some random element there, then we likely have some kind of a bug that we should track down.

RI0, 2016-03-05:
OK - Noted.
Have dealt with string tension and will video tests on the inking - last time it did not
do it at the same place. Will make sure the laptop has the latest version of RoboPaint. 
We also tried to run the print on top of the previous painting to see if they lined up - the 
variance in the layers is part of what led to the Serendipity conversation.

RI0, 2016-03-10:
We tightened the strings and reinstalled RoboPaint. Prints are 
consistent and re-inking takes place at the same time with 
each print ( tested with 3 runs on the same page ) 
Not sure what was happening but it does not seem to be an 
issue now. Thank you.

rogerfachini, 2016-03-15:
@RI0 If you didn't already know, last year for Robogames I made a fully generative app that takes audio input (file, microphone, etc) and makes a spiral pattern from it in real time. I wrote it in python, but I can start porting it over to JS as a mode if you guys think it could be useful

RI0, 2016-03-15:
@rogerfachini - I did not know that. That sounds really exciting. 
My little sister Sofia is entering the RobotArt competition with 
her class. 

A generative mode in RoboPaint would be great. Have you ever 
thought about using Perlin Noise.

PERLIN NOISE - CLOUD EXAMPLE
http://www.noisemachine.com/talk1/24b.html 

ANOTHER EXAMPLE
http://josephg.github.io/noisejs/demo3d.html
https://github.com/josephg/noisejs

RI0, 2016-03-16:
Or maybe there are items here in JS that could be useful

TUTORIALS
http://p5js.org/tutorials/

EXAMPLE WAVE - with Noise 
http://p5js.org/examples/examples/Math_Noise_Wave.php

EXAMPLE BRUSH "Error" using Noise
http://p5js.org/examples/examples/Math__Noise1D.php

docprofsky, 2016-09-28:
If this is implemented it probably should be done as a separate mode. I have added this mode idea to #193.
https://github.com/evil-mad/robopaint/issues/193#issuecomment-250064913

closed 2016-09-28 by docprofsky