evil-mad/robopaint #260: RoboPaint: Beta 2 Release checklist

techninja, 2016-04-01:
Some of the biggest improvements to come in the latest Beta1 release have now been vetted and shown to work fairly well, but those that haven't have been somewhat prioritized and will need to be fixed in the forthcoming Beta2 release.

This issue is to discuss exactly what those issues are and prioritize them over other things to get the Beta2 release out sooner rather than later, so we can be one step closer to the "next generation" 2.0 non beta release.

The following is a living checklist of items to be discussed below. When all of them are checked off (or prioritized out), beta 2 should be ready to release.
- [ ] CNC Server performance branch:
  - This completely changes the way in which CNC Server works to use a multi-process system, and has been in progress since the discovery of the wait times to "prefill the buffer" for larger sets of instructions.
  - This has already progressed forward enough to prove that it speeds up the entire system to where we no longer need the prefill, and is incredibly reliable, but it changes so much that it will have to be tested and retested, and anything touching the buffer will have to be rethought a bit.
 - When my (and @docprofsky's) testing on this is done I'll roll a new feature breaking release of CNCserver and publish it on NPM, and we can assign that version lock to RP
- [ ] UI/UX improvements (up for discussion)
  - [ ] _Improve home screen:_ Parts of the new home screen have been a success, others (like the inability to rely on the position of any home screen bubble) have not.
  - [ ] _Verify settings screen option placement:_ A number of "home locations" for settings were adjusted as an idea to test, and haven't possibly been as intuitive as we'd originally hoped.
  - [ ] _Solution for too many icons in the toolbar:_ In dev builds this is all over the place and needs a solution as the number of awesome mode ideas grows.
- [ ] Default modes: Spiral mode is fun, but I'm not sure it should be a default mode given the lack of clear usage instructions. I'm also not sure that where the modes selector is (hidden in the settings window) makes sense, perhaps a new modal selection window teased from some kind of "Enable/Change modes" icon & text on bottom right corner of the home screen?
- [ ] #252 - SVG history feature as described in the issue [here](https://github.com/evil-mad/robopaint/issues/252#issuecomment-173060764).

It's possible I could get in the stipple mode for the next beta as it was _**mostly**_ just waiting for the performance branch of cncserver to hit. 

I think that's about it? Looking for input from @docprofsky, @oskay, @rubyjlp, @rogerfachini or anyone else on what remaining important issues for users there are.

oskay, 2016-04-01:
Great-- excited about this!

#252 & #253 are definitely at the top of my list. 

- I would make Spiral _not_ a default mode.

techninja, 2016-04-02:
Awesome! It should be noted then that the fix for #253 has already been implemented in c6deca82c. This would allow people with node.js installed locally to simply check a box and be freed from any background slowdowns that could effect running the buffer. Admittedly this would still effect the buffer filler, but I think we can't know for sure how far reaching the effect is till we can reproduce the circumstances with a new build and see how it reacts with both methods :)

oskay, 2016-04-02:
(It would be much, much better to have this in a build, such that people who just _download_ RoboPaint could have it working at reasonable speeds, too.)

techninja, 2016-04-02:
It will. I think you misunderstand that. The new cncserver runs multiprocess in an electron webview by default, and the option shown above is for running it in a completely separate node.js process. The first one doesn't requir eyou to do anything different in running or installation and runs great, the second allows you to get past the "minimized in the background" performance issue that electron has by running the cncserver runner process completely outside of electron :smile:

oskay, 2016-04-02:
Then confirmed that I don't understand. It sounded like you said that this was for people who have node.js installed.

techninja, 2016-04-02:
The **_background performance_** issue fix is for people with node.js installed who click the checkbox. The full perfomance stuff you get regardless. I just printed a full Mona Lisa at 100% speed (no paint) from opening and pressing the button to done in ~3min. There is now basically zero time from when the data is built to when it's printing, and it spools faster than the bot can print at full speed, exactly what we wanted.

rubyjlp, 2016-04-02:
In regards to having Spiral Paint not being a default mode -
I actually like having it as a default. It's a really cool functionality of the program and I think newcomers would want to see what kinds of capabilities it has. @oskay, I think I recall you saying on a different thread that you thought it didn't have good enough instructions. Personally, I found it very simple to use, but if you'd like I could write a set of instructions for it on the wiki.

oskay, 2016-04-03:
I welcome differing opinions! If you're willing to help out with the documentation-- to make it absolutely clear to a novice how to use it _well_, then I'm happy to have it on the default screen.

techninja, 2016-04-12:
CNCServer Scratch tests are now all passing beautifully. Required a bit more rework, but the good news is that running a child process doesn't effect the other child processes, so when running the test, it suddenly has grandchildren! Also added a new command line option to just show serial data, so if you want to see serial, you don't have to see debug stuff, and vice-versa.

I'd really like to get in some part of the acceleration work that @docprofsky started, though I suspect it's high on the "not ready for prime time" scale, and a bit more thought needs to go into it before we just go ramping up and down every movement. Also everything works fine without it at the moment.

I'm going to check off the box for CNCServer, though not actually release it yet as I'm sure I'm likely to find a few more bugs. There's been a great sea-change over at node-serialport HQ, and a new version release that messes with all my current build release file placement ideas, but you can build it quite easily compared to the current hassle.

Going to switch over to #252, then circle back to make sure n-sp 2.1.0 works like I hope it will across the different systems.

Bonus: I've been using the stipple mode as test. If there's over 10k stipples it lags a bit during the initial send, but whips right out and starts drawing. Assuming I can get it running happily on all the systems, I'll include it in the next beta. Any suggestions for Icon?

oskay, 2016-04-12:
This all sounds great.

Excited to hear about better performance from Scratch, especially. 

I would again recommend acceleration for pen-up movements only, at this stage-- if at all. I can help with fine tuning that code if needed.

How about a close-up of a stippled drawing for the icon?

techninja, 2016-04-12:
Seeing as Schuyler is off doing FRC, I don't see accel stuff getting in till after beta2 anyways.  Icon idea is doable.

slimline33, 2016-07-19:
Hi techninja,

I make a diy plotter and I think about to use cnc-server on a raspberry for stand alone usage. 

I love your software robopaint. I wish the possibility to make a virtual device and connect to the cnc server by a ip number or something. Then anybody can use your robopaint for all cnc/drawing machines :)

techninja, 2016-07-19:
@slimline33 Hello! In the soon to be released beta we have experimental remote IP address cncserver support! Certain parts about it are a little broken, specifically machine selection, but it does mostly work. If you've made your own plotter, I'm assuming using the [EBB](http://www.schmalzhaus.com/EBB/), you should be able to make a new ini configuration file and run this on the Raspberry Pi. Though I've experimented successfully with running custom machines using Arduinos and GCODE. We'll have the release out quite soon, feel free and open an issue if you're having specific problems, or if you want to discuss how to get it working, join us in our [Gitter chat room](https://gitter.im/evil-mad/robopaint) :)

slimline33, 2016-07-19:
@techninja Great. I am currently in the planning stage and I'm looking for a very very simple solution. At the moment I will use the EBB. However, an Arduino would not be bad. Especially as regards the extensibility! I would prefer even a tool change (different pens) and the EBB have only 2 motor outputs and one servo out. And I'm not familar with the ic ... but currently still dreaming.

Once it starts, I'm certainly times in your chat room! Your support is fantastic!

EmbeddedMan, 2016-07-19:
Don't forget that the EBB can easily support up to 8 RC servos, so you
could build a tool changer with several servos using the EBB.

*Brian

slimline33, 2016-07-19:
@EmbeddedMan ! Oh. Thank you very much. that was new to me.

techninja, 2016-07-30:
Unfortunate we didn't get #242 in for this, but the timeline just didn't have it. Beta 2 release was locked and tagged for Friday, July 29 2016 (my 33rd Birthday!), should be out of draft and published by Monday or Tuesday night.

closed 2016-07-30 by techninja