evil-mad/robopaint #262: Extreme slowness when tracing certain files

oskay, 2016-06-10:
Attached are two example files where RoboPaint seems to hang in the tracing stage. One of them is a very simple file, but still after 1 hour is in tracing (see screenshot).

[image]

[RoboPaintFails.zip](https://github.com/evil-mad/robopaint/files/309426/RoboPaintFails.zip)

docprofsky, 2016-06-13:
This is half fixed/bypassed.
evil-mad/robopaint@b3374fa fixes the problem with `nothing after 1 hour.svg`, but `nothing.svg` is still breaking RP.

Each file is causing problems in different places.
`nothing after 1 hour.svg` here https://github.com/evil-mad/robopaint/blob/e3f8e5b7d13cc2ccd6ef7a15bdee2e32175e4a20/resources/rp_modules/paper.auto.fill.js#L757
`nothing.svg` here https://github.com/evil-mad/robopaint/blob/bcebe2e23e8b22dfe51b50a6b7eff0c77ec8cb6f/resources/rp_modules/paper.auto.stroke.js#L313

Additional information can be found here.
https://gitter.im/evil-mad/robopaint?at=575e16a0e42570166a84616d

docprofsky, 2016-07-16:
I have done more testing on these files and have added some more fixes for them. They are in the flatten-fix branch. I believe I have found all of the problems caused by the files for all except the overlay fill types.

Should I open a PR for flatten-fix?

Reff99, 2016-07-17:
Hello - I'm the person with the problem, who first sent the files. As a relatively non-technical person, can someone tell me if there is now an update which lets me now paint these and other files (these were just two examples: I couldn't reproduce nearly every file I tried). Thanks!

techninja, 2016-07-17:
Once testing is done and the remaining fixes has been merged we should be in good shape to roll there next beta release _very soon_. @docprofsky has been finding and fixing bugs so it's good we're sticking with our current plan.

Reff99, 2016-07-17:
Ok, thanks a lot
Brian
>

docprofsky, 2016-07-17:
@Reff99 Can you run the `flatten-fix` branch and see if you still get any issues with it?

oskay, 2016-07-17:
@docprofsky: If you're going to ask a question like that, you should provide explicit instructions for how to test it. For example, is there a compiled version available to test somewhere? Or, if an existing RoboPaint install can be modified by replacing certain files, please describe how to do so.

ajpiano, 2016-07-18:
With @docprofsky's branch and help I was able to get "Nothing after 1 hr.svg" to print successfully and also got some other pieces to trace and begin printing more or less "successfully" as well :) Seems like this is... better than what's there !

closed 2016-07-19 by techninja

oskay, 2016-08-30:
Fantastic-- thank you for following up!

PR #270 (2016-07-19): Added more changes for #262 (flatten-fix)
These changes help fix #262.
Should I do this before all calls to flatten?