evil-mad/robopaint #275: Some SVG files appear to be all black in color

oskay, 2016-08-09:
[image]

Some SVG files, when opened in RoboPaint, show the correct shapes, but filled with solid black color. As this appears in the edit mode, where the file can be imported, it has consequences for using the file elsewhere in RoboPaint as well.

_Background:_ There are quite a few different methods of indicating what the styles of objects are within an SVG file, including tags on the individual objects, inherited tags, and css, both inline and by named style. (Perusing the SVG export options within Adobe Illustrator is somewhat instructive about the variety of different methods available.) It seems likely that some, but not all, of these methods are not being understood/interpreted correctly.

_Information needed:_ To proceed with understanding and fixing this issue, we need a set of SVG files that can reproduce the problem. If you know of one or more, please ZIP the file and attach it to this issue, ideally with a screen shot as well.

We believe that the following are true:
* The choice of machine does not affect this behavior.
* Some, not all, SVG files are affected.

Questions to be answered:
* Are only some computers/platforms affected?
* Does the file still _print_ correctly in multiple colors, even if it appears all black on the screen?
* Do the affected files have a consistency about how their styling is applied?
* Were the affected files generated by the same software?

RI0, 2016-08-10:
I had the same problem with files created in Adobe Illustrator. 
Sometimes it was possible to flatten the image or export individual
layers and then re-layer the images in RoboPaint.

oskay, 2016-08-10:
If you can provide an example for us to work with, that would be helpful. :)

RI0, 2016-08-11:
THIS ONE WAS FINE

<svg width="11in" height="7.697917in" xmlns="http://www.w3.org/2000/svg">
 <g>
  <g id="svg_1">
   <path fill="#ff0000" stroke-width="0" id="svg_2" d="m617.637695,130.172241c8.200012,-8.500008 7.700012,-9.200005 19.500061,-9.200005c11.799927,0 11.799927,14.299995 23.699951,14.299995c11.799988,0 11.799988,-18.099998 23.700012,-18.099998c11.800049,0 11.800049,4.199997 23.700012,4.199997c11.799988,0 11.799988,-2.599998 23.700012,-2.599998c11.799988,0 11.799988,12.800003 23.699951,12.800003c11.800049,0 11.800049,2.699997 23.700012,2.699997c11.800049,0 11.800049,-15.799995 23.700012,-15.799995s17.799988,-3.200005 26,5.299995c8.5,8.800003 -15.700012,14.5 -15.700012,26.59996c0,12.199982 8.200012,12.199982 8.200012,24.399979c0,12.200027 16.299988,12.200027 16.299988,24.400009c0,12.199997 -23.899963,12.199997 -23.899963,24.399994c0,12.199982 9.5,12.199982 9.5,24.399994c0,12.199982 -1.799988,12.199982 -1.799988,24.399994c0,12.200073 13.599976,12.200073 13.599976,24.400085c0,12.199982 -8,12.199982 -8,24.399994c0,12.200012 -2.200012,12.200012 -2.200012,24.399963c0,12.199982 13,12.199982 13,24.399963c0,12.200012 -13,12.200012 -13,24.400024c0,12.199982 4.600037,12.199982 4.600037,24.399994c0,12.199982 -14.5,12.199982 -14.5,24.399994c0,12.200012 21.399963,12.200012 21.399963,24.399963c0,12.200012 -2.899963,12.200012 -2.899963,24.399994c0,12.200012 -12.400024,12.200012 -12.400024,24.399994c0,12.200012 -8.5,12.200012 -8.5,24.39978c0,12.200012 23.799988,12.200012 23.799988,24.400024c0,12.199951 -16,12.199951 -16,24.399963c0,12.200012 15.799988,16.700012 7.299988,25.5c-8.199951,8.5 -13,11 -24.799988,11c-11.799988,0 -11.799988,2.400024 -23.700012,2.400024c-11.799988,0 -11.799988,-11.099976 -23.700012,-11.099976c-11.799988,0 -11.799988,-9.300049 -23.699951,-9.300049c-11.799988,0 -11.799988,18.600037 -23.700012,18.600037c-11.799988,0 -11.799988,-0.900024 -23.700012,-0.900024c-11.799988,0 -11.799988,-9.099976 -23.700012,-9.099976c-11.799988,0 -11.799988,-12.400024 -23.699951,-12.400024s-9.700073,12.700012 -17.900024,4.200012c-8.5,-8.799988 2.799988,-6.599976 2.799988,-18.799988c0,-12.200012 -13,-12.200012 -13,-24.400024c0,-12.200012 -3.200012,-12.200012 -3.200012,-24.400024c0,-12.199768 11.5,-12.199768 11.5,-24.39978c0,-12.200012 3.900024,-12.200012 3.900024,-24.399963c0,-12.199982 -10.700012,-12.199982 -10.700012,-24.399994c0,-12.200012 5.399963,-12.200012 5.399963,-24.399994c0,-12.199982 -10.399963,-12.199982 -10.399963,-24.399963c0,-12.200012 4.400024,-12.200012 4.400024,-24.400024c0,-12.199982 5.599976,-12.199982 5.599976,-24.399994c0,-12.199982 -9.700012,-12.199982 -9.700012,-24.399994c0,-12.200012 -0.399963,-12.200012 -0.399963,-24.399994c0,-12.199951 5.5,-12.199951 5.5,-24.399933c0,-12.200012 14.399963,-12.200012 14.399963,-24.400116c0,-12.199982 -21.5,-12.199982 -21.5,-24.399994c0,-12.200012 7.300049,-12.200012 7.300049,-24.400009c0,-12.199982 -7.400024,-12.199982 -7.400024,-24.399994c0,-12.199997 6,-12.199997 6,-24.400009c0,-12.199997 -8.599976,-10.599991 -10.100037,-22.799973c-1.099976,-10.699982 9.500061,-13.09996 18,-21.899963l0,0.000015l0,0.000015l0,0.000015z" class="st0"/>
  </g>
  <g stroke="null" id="svg_26">
   <g stroke="null" transform="matrix(0.996172 0 0 0.996172 -199.248 -217.408)" id="svg_24">
    <path fill="#0000ff" stroke="null" id="svg_25" d="m374.874603,348.730042c7.899933,-8.400024 10,12.199951 21.5,12.199951c11.5,0 11.5,-12.599976 23,-12.599976c11.5,0 11.5,-12.700012 23,-12.700012c11.5,0 11.5,21.799988 23,21.799988c11.5,0 11.5,4.200073 23,4.200073c11.5,0 11.5,-9.300049 23,-9.300049c11.500214,0 11.500214,2.599976 23.000397,2.599976c11.500244,0 11.500244,-0.599976 23.000427,-0.599976s25.600464,-25.299988 33.400391,-16.900024c8.400024,8.900024 -23.100281,22.099976 -23.100281,34.300049c0,12.199951 25.200378,12.199951 25.200378,24.400024c0,12.199951 -13.700134,12.199951 -13.700134,24.399902c0,12.200073 -6.100159,12.200073 -6.100159,24.400024c0,12.200073 19.400269,12.200073 19.400269,24.400024c0,12.199951 -26.800415,12.199951 -26.800415,24.399933c0,12.199951 2,12.199951 2,24.400177c0,12.200195 -0.599976,12.200195 -0.599976,24.400269c0,12.200073 19.000366,12.200073 19.000366,24.400024c0,12.200012 -11.100159,12.200012 -11.100159,24.399902c0,12.19989 -9.100159,12.19989 -9.100159,24.399963c0,12.199951 13.800171,12.199951 13.800171,24.400024c0,12.199951 -1,12.199951 -1,24.399902c0,12.200073 -10.500183,12.200073 -10.500183,24.400024c0,12.200073 19.700195,12.200073 19.700195,24.400024c0,12.199951 1.699951,12.199951 1.699951,24.400024c0,12.199951 3.400024,12.199951 3.400024,24.400024c0,12.199951 -12.999939,12.199951 -12.999939,24.399902c0,12.200073 9.299927,12.200073 9.299927,24.400024c0,12.200073 5.100098,20.900024 -3.299988,29.799927c-7.900024,8.400146 -17.200256,8.100098 -28.7005,8.100098c-11.500183,0 -11.500183,-21.300049 -23.000427,-21.300049c-11.500183,0 -11.500183,16.300049 -23.000397,16.300049c-11.5,0 -11.5,-0.400024 -23,-0.400024c-11.499939,0 -11.499939,4.099976 -22.999939,4.099976c-11.5,0 -11.5,-16.899902 -23,-16.899902c-11.5,0 -11.5,-9.200073 -23,-9.200073c-11.5,0 -11.5,23.600098 -23,23.600098s-13.099976,-3.200073 -21,-11.500122c-8.399902,-8.899902 -12.899902,-10.399902 -12.899902,-22.599976c0,-12.199951 10.399902,-12.199951 10.399902,-24.400024c0,-12.199951 -10,-12.199951 -10,-24.399902c0,-12.200073 11.5,-12.200073 11.5,-24.400024c0,-12.200073 -15.199982,-12.200073 -15.199982,-24.399963c0,-12.19989 24.300079,-12.19989 24.300079,-24.399902c0,-12.199951 -1.300079,-12.199951 - [... cut ...] 5 -1,7.899902 1.199951,8.699951zm-23.300049,5.5c3.400024,0.699951 9.900024,0.900024 11,-4.699951c-5,1.099976 -13.400024,-5.700073 -15.599976,-0.200073c2.200073,-0.199951 5,2.5 4.599976,4.900024zm-3.199951,13.699951c-0.199951,-1.599976 4.199951,-8.799927 0.599976,-10c0.099976,1.600098 -4.299927,8.800049 -0.599976,10zm-24.400024,30.300049c2.099976,0.300049 7.800049,-6.699951 4.300049,-8c-0.200073,2.199951 -4.700073,5.199951 -4.300049,8z" class="st2"/>
     </g>
    </g>
   </g>
  </g>
 </g>
</svg>

RI0, 2016-08-11:
It looked like Illustrator added extra header information and it 
created a style sheet in the SVG. 

`<?xml version="1.0" encoding="utf-8"?>

<svg version="1.1" id="Layer_1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" x="0px" y="0px"
	 viewBox="0 0 841.9 595.3" style="enable-background:new 0 0 841.9 595.3;" xml:space="preserve">
<style type="text/css">
	.st0{fill:#FFFFFF;}
	.st1{fill:#ED2224;}
	.st2{fill:#37429A;}
	.st3{fill:#010101;}
</style>`

We removed all of this and made it look more like an SVG generated by RoboPaint. 

We created a simple drawing in RoboPaint with the colours I wanted to use and then 
used the colour codes to modify the SVG exported from Illustrator. 

As mentioned above, we also tried importing individual shapes into RoboPaint, changing 
the colours and then saving the file as an SVG. This sometimes would produce a clean 
SVG that would render in the print section of RoboPaint. 

Most of the time it is fine in edit mode but then turns black in the print/paint mode
if the SVG is not clean.
 
Hope this helps

RI0, 2016-08-11:
Since Github will not accept an SVG the extension was changed to txt
[PPT.txt](https://github.com/evil-mad/robopaint/files/414221/PPT.txt)

oskay, 2016-08-11:
I would recommend using a zip archive for SVG files; they can be attached here, and it's a way to prevent corruption by browsers during the upload/download process.

ajpiano, 2016-08-30:
It seems the difference is rooted in one of the export options from Illustrator; selecting "Style Attributes" results in RoboPaint being able to deal with the colors, "Style Elements" results in all black in RoboPaint, as you can see in this screenshot where I imported each svg into robopaint.
[image]
I have attached [this zip file with both files in it for y'all to review](https://github.com/evil-mad/robopaint/files/446096/cssconf-robopaint-illustrator-svg-export-modes.zip)

oskay, 2016-08-30:
Interesting; I have seen various errors with AI SVG export (with other options) before. Thank you for labeling these files clearly!

One possible approach (depending on where the offending SVG parsing code lies) would be to give the user a warning, and some words about how to properly export from AI. However, this is a little unkind, since many people will be working from artwork of unknown origin. Another approach would be to parse the elements upon opening the file, and apply those styles as attributes instead.

oskay, 2018-11-16:
See: https://github.com/evil-mad/robopaint/issues/275#issuecomment-243611442

kritikaperiwal12, 2019-02-27:
Hello,
While showing SVG file on view using Pocket SVG Framework it is showing black color filled. Please tell me the reason behind it.

arcovirtual, 2019-02-27:
I solved it and I created a mod for opencart https://www.opencart.com/index.php?route=marketplace/extension/info&member_token=qC6vrvaV71qxSuUggfC1gRL28UKkwXH7&extension_id=35539 regards

ammar23, 2019-08-16:
first, It must be tag style such as   ' <style type="text/css"> '
second, Resave file SVG on Adobe xd and export as SVG

docsmooth, 2021-02-02:
hacky fix for the original files: https://github.com/docsmooth/fix-svg