evil-mad/robopaint #281: No build for Linux x86_64

scruss, 2016-10-11:
### Description

Robopaint will not run on 64 bit systems as the binaries are all 32 bit.

#### Steps to reproduce OR Feature idea (Delete one)
  1. extract the archive
  2. double click on the icon
  3. … nothing happens
  4. open a terminal, execute the program, and get `error while loading shared libraries: libgobject-2.0.so.0: cannot open shared object file: No such file or directory`
  5. check the executable with `file` and find that it's 32 bit only.

### Version affected (if bug)
v0.9.5

#### What operating system are you running?
Linux: Ubuntu 16.04 x86_64

#### How did you install RoboPaint?
Copied files from zip

#### What robot are you using?
Other (specify): Roland plotter

#### Is the robot you want to control directly connected to RoboPaint?
No, I am using RoboPaint in simulation mode.

oskay, 2016-10-11:
Have you tried the newer versions? I would recommend trying with v2.0.0 Beta 2.

scruss, 2016-10-11:
I followed the *[Click here for the latest release](https://github.com/evil-mad/robopaint/releases/tag/v0.9.5-2015.01.25)* link under [Downloads / Install](https://github.com/evil-mad/robopaint#downloads--install "Downloads / Install"), and it took me to v0.95

docprofsky, 2016-10-11:
v0.9.5 is the latest stable release, however the v2.0.0-Beta.2 (and Beta.1) release have many significant improvements to RoboPaint. We probably should update the documentation to reflect this.

You can find the v2.0.0-Beta.2 release on the [releases page](https://github.com/evil-mad/robopaint/releases).

oskay, 2016-10-11:
It does indeed link to the older version. :(  
This does appear to be a significant failure of documentation.

Would you please give the current version a try?

docprofsky, 2016-10-11:
It links to the most recent release, ignoring pre-releases.