evil-mad/robopaint #288: Painting is 1 inch off

pelikhan, 2016-12-16:
### Description

Everything works fine with 0.9. However with 2.0beta3, the painter starts with a 1 inch offset in x and y. It misses the second water holder and picks the wrong color. Reseting the painter does not solve the problem. Note that I had to install 0.9 to get the drivers.

#### Steps to reproduce OR Feature idea (Choose one)

1. Install 2.0beta3
2. Try to print.

#### Version affected (if bug)

2.0 beta3

#### What operating system are you running?

Windows 10.

#### How did you install RoboPaint?

System installer from 2.0beta3 release on this repo

#### What robot are you using?

WaterColorBot

#### Is the robot you want to control directly connected to RoboPaint?

Yes, the robot is plugged into the computer I am using.

oskay, 2016-12-16:
When you say "Reseting the painter," can you please clarify what you are referring to?

And, if you try to print the same thing twice, is the position of painting the same both times?

pelikhan, 2016-12-16:
> Reseting the painter

unplug both cables, park home motor, replug both cables, close/launch robotpaint again.

> if you try to print the same thing twice, is the position of painting the same both times?

Yes, the offset error is consistent. Roughly 1 inch in both directions. I can hear the motor skipping when trying to more the brush beyond the allowed distance. I can provide video if needed.

docprofsky, 2016-12-16:
After the motors skip do they skip again?
Does this happen with one of the examples?

This seems like it might be an issue with the CNCServer WaterColorBot machine definition. RoboPaint should not be able to command a move outside the robot bounds.

oskay, 2016-12-16:
Does it return to home properly after finishing?

oskay, 2016-12-16:
Also: Can you please say what driver you needed? The WaterColorBot does not need a separate driver in Windows 10.

pelikhan, 2016-12-17:
I first installed 2.0beta3 and the printer was not recognized by robopaint. So I installed 0.9.5, which installs a driver, and it started working.

pelikhan, 2016-12-17:
No, it sits within 1 inch of proper home location.

oskay, 2016-12-17:
It sounds like all of the positions are off then-- that suggests that there may be a mechanical problem-- this may be independent of RoboPaint.  Does the same issue occur at lower speeds?  And, have you tried to use the Inkscape software to see if things are working there?

pelikhan, 2016-12-17:
It print correctly in RoboPaint 0.9.5.

pelikhan, 2016-12-17:
The InkScape extension v1.1.0 worked fine.

oskay, 2016-12-21:
Did you re-install the current version after going back to 0.9.5?