evil-mad/robopaint #289: After installation of 2.0, RobotPaint cannot be found on Windows

pelikhan, 2016-12-16:
### Description

No App menu entry, desktop icon to relaunch RoboPaint after installation.

#### Steps to reproduce OR Feature idea (Choose one)
  1. Run installer
  2. Close robopaint
  3. Try to run again.

#### Version affected (if bug)

2.0beta3

#### What operating system are you running?
Windows: 10

#### How did you install RoboPaint?
System installer

#### What robot are you using?
WaterColorBot

#### Is the robot you want to control directly connected to RoboPaint?
Yes, the robot is plugged into the computer I am using.

oskay, 2016-12-16:
Can you please specify what the failure mode is? You say "Try to run again", but don't say what it is that you're actually doing, nor what the behavior is when you do that.

pelikhan, 2016-12-16:
Ha it does work now of course. Open start menu and search for "robopaint". Sorry for the confusion.

closed 2016-12-16 by pelikhan