evil-mad/robopaint #292: Running from Docker container

jayfresh, 2017-04-24:
### Description

I'd like to run this in a Docker container, mainly to avoid the Visual Studio dependency on my Windows host. Has anyone done this? It looks like there is some precedent for running electron apps in docker containers: https://medium.com/@calbertts/developing-electron-apps-in-macos-afd21b4a59e3 and https://blog.jessfraz.com/post/docker-containers-on-the-desktop/

#### Feature idea
This new feature would be a Dockerfile (or maybe a Docker compose file) that I can run to load robopaint in a Docker container.

#### What operating system are you running?
Windows: 10

#### What robot are you using?
AxiDraw

#### Is the robot you want to control directly connected to RoboPaint?
Yes, the robot is plugged into the computer I am using.

oskay, 2017-04-24:
@techninja Is there actually a Visual Studio dependency?

oskay, 2017-04-24:
@jayfresh I have not used Docker, or even genuinely understand its purpose. Perhaps you can clarify what your end goal is, so that we can provide further pointers to the correct resources?

I will note that I use RoboPaint on Mac, Windows, and Linux, and have never installed Visual Studio. It seems like there should be some solution here.

techninja, 2017-04-24:
Sorry, at a conference right now. Visual Studio is only required for building node-serialport on system that node-pre-gyp doesn't account for when _**developing**_, not for running the compiled release version. I've found that running from node 4.2.0 is reliable, but others have caused problems. Docker is a container system for virtual machines, though even I am a little unclear about how it would be used. I'm personally finding coming back to the project that Windows has its own share of oddities in running the current RP & cncserver versions that are _not ok_ (specifically relating to unpredictable plots). I also suspect that recent changes in windows 10 point releases are causing issues. Hopefully with a little more information on exactly what @jayfresh is looking to do, we can figure htis out.