evil-mad/robopaint #295: v 2.1.0 (Mac) shows as damaged

oskay, 2017-12-05:
Attempting to open this on the Mac shows the infamous "this file is damaged" dialog, which usually indicates that the application is not properly signed.

techninja, 2017-12-05:
Weird! I'm getting a new macbook charger very soon so I should be able to verify, it's possible the upload failed, but I'm pretty sure nothing funny has happened to the signing to mess this up in the intervening months. Will check tonight.

techninja, 2017-12-06:
El Capitan, downloaded fresh DMG, extracted and installed with no issues. App runs fine. I guess time to upgrade the OS again? 😕

oskay, 2017-12-06:
Yes, I think that's it. I believe that you need to be on Sierra or later to sign currently-accepted packages.

techninja, 2017-12-06:
Oh sheesh. When the charger comes I'll get the upgrade going

techninja, 2017-12-07:
OK! So.. apparently Apple is a terrible company and hates loyal people who appreciate their hardware. My only macbook laptop from ~2009 is now on the no longer supported list, and will be arbitrarily denied OS software updates now, even though it seems to run beautifully on el capitan. 

https://www.intego.com/mac-security-blog/what-to-do-if-your-mac-cant-run-macos-sierra/

...now what?

techninja, 2017-12-07:
I mean this with full sincerity: As long as your local XCode is updated with the correct certs, building, signing and making releases is vastly more tame than it has been in the past. Lemme know what options you think we've got and we can chat. Maybe you've got an older yet newer macbook laying around 😆

oskay, 2017-12-07:
Well... that computer *is* 8 years old....

I don't have any more recent computers lying around (recently recycled a bunch). It may be time to temporarily go back to having me sign the packages.

docprofsky, 2017-12-07:
I am on 10.10, Yosemite, though I do not think this can help. I also have access to a PPC iMac.

EmbeddedMan, 2017-12-07:
You can easily make a Mac in VirtualBox, and run the latest OS that way on your Windows or Linux machine. Would that help?

techninja, 2017-12-07:
Yes, I suppose it will! I suppose I'll head down the rabbit hole. 🕳🐇

techninja, 2017-12-08:
[image]

UM... Might this be the problem?

oskay, 2017-12-08:
Yup. Looks like you might need to renew that. :P 

The one on my computer is good to 2022.

oskay, 2017-12-08:
Looks like the certificate in your name (on our account) is valid to Dec 07, 2018.

techninja, 2017-12-08:
I think what matters here is that cert. I've downloaded the new one, but it isn't available as a signing identity. I sure love clear as mud this always is. Thanks apple :rage3: 🍎  💥

techninja, 2017-12-12:
Aha. Lucky me! [Upgrade is in fact possible.](https://www.youtube.com/watch?v=tZ6TomIhqcY)

techninja, 2017-12-15:
FWIW: I've upgraded!... but I've yet to find the time to properly fix the signing cert, because as simple as you'd think it should be.. it is not. 😖

oskay, 2017-12-16:
No, you wouldn't think it's simple. Having been through this a few times, I still need to look through the docs every time: https://developer.apple.com/library/content/documentation/IDEs/Conceptual/AppDistributionGuide/MaintainingCertificates/MaintainingCertificates.html

lalo, 2018-01-08:
Any update on this issue?

techninja, 2018-01-22:
@lalo Et Al: Sorry for the wait, life has been crazy as usual. I have my upgraded mac running, I hope to take another few hours soon and simply reset my local certificates (or try under a new user account) to get this working ASAP. Will likely get this out as RP 2.2.0 and a few other bugfixes and upgrades very hopefully before the end of the month.

techninja, 2018-01-31:
Ok! I wouldn't believe it if I hadn't seen it myself, but I made a new mac appstore 3rd party distribution cert from this machine (something I was planning on doing anyway to get this on the mac appstore!), and I think it worked!!

@oskay Please test [this testing build](http://robopaint.tn42.com/RoboPaint_Mac_v2.1.0.dmg). If this works, then I can focus on wrapping up the rest of what's needed for this next release (bugfixes and security updates). WHEW.

oskay, 2018-02-06:
I downloaded it and opened it. It gives a warning that it is from an "unidentified developer", but if you give it permission, it does open and connect.  

Is that using an Evil Mad Scientist certificate, or a different one?

techninja, 2018-02-08:
@oskay Sorry for the wait, just got back from a work trip. That is strange that it says it's from an unidentified developer, this was signed with the new 3rd party distribution certificate, and gave no error. I'll run some diagnostics and attempt more verification as this should be cinched.

oskay, 2018-02-08:
It's a partial success-- current MacOS will refuse to open any _unsigned_ package....

techninja, 2018-02-08:
...huh. These are the two command suggested by apple to verify signatures.

```
shadow:build techninja$ codesign --verify --deep --strict --verbose=2 dist/RoboPaint.app/
--prepared:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/Mantle.framework/Versions/Current/.
--validated:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/Mantle.framework/Versions/Current/.
--prepared:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/ReactiveCocoa.framework/Versions/Current/.
--validated:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/ReactiveCocoa.framework/Versions/Current/.
--prepared:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/RoboPaint Helper EH.app
--validated:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/RoboPaint Helper EH.app
--prepared:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/RoboPaint Helper NP.app
--validated:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/RoboPaint Helper NP.app
--prepared:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/RoboPaint Helper.app
--validated:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/RoboPaint Helper.app
--prepared:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/Squirrel.framework/Versions/Current/.
--validated:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/Squirrel.framework/Versions/Current/.
--prepared:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/Electron Framework.framework/Versions/Current/.
--validated:/Users/techninja/web/robopaint/code/build/dist/RoboPaint.app/Contents/Frameworks/Electron Framework.framework/Versions/Current/.
dist/RoboPaint.app/: valid on disk
dist/RoboPaint.app/: satisfies its Designated Requirement
shadow:build techninja$ spctl -a -t exec -vv dist/RoboPaint.app/
dist/RoboPaint.app/: rejected
origin=3rd Party Mac Developer Application: Evil Mad Science LLC (67VYYJK96Z)
```

Sooo, it verifies and it doesn't. Lovely. More digging is needed I guess 😢

techninja, 2018-03-07:
@oskay Spent 2 hours on this. Learning:
- Developer ID application certs are restricted to the private keys/certificate signing request of the specific machine they were created on. When I upgraded, or when it expired, these broke for my local machine, and though I can download/install the Developer ID cert, it's **not** a valid signing identity because it's missing the private key.
- Only "Agents" (of which _there can be only **one**_) are allowed to create these, hence the insanity we had to jump through to get this working last time to get a private key from you.
- Developer ID certificate signing is really supposed to only be used for "development" level distribution, which. I think is wrong, so..
- Been working through [this incredible Mac App Store signing guide](https://mintkit.net/electron-userland/electron-osx-sign/guide/) to use the app id, distribution provisioning profile and distribution cert I made last time. This has yet to work in a way that I can call ready for testing, but will post back here ASAP if I get something promising.
- Sorry for the hassle and wait 😢

oskay, 2018-04-30:
Any progress on this? There is a new batch of WaterColorBots starting to ship. ;)

derrallg, 2018-05-15:
The testing build posted on January 30th worked for me. I was also getting the Damaged File prompt with the other version. I thought it was odd that the earlier version wasn't available to approve for download in the Security settings and with the update I was able to approve.

jrheard, 2018-10-18:
FYI: i see this "damaged file" behavior when i try to run robopaint v 2.1.0 on mac os x version 10.14, on a macbook air.

oskay, 2018-10-18:
Yes, it's the signature.

lalo, 2018-10-20:
I wonder if this might help https://developer.apple.com/developer-id/

oskay, 2018-10-20:
We are aware of the method -- it's been signed in the past -- but there has been some kind of an issue with the setup on @techninja's computer.

oskay, 2018-10-20:
Since this still seems to be an issue, I've gone ahead and signed a copy of v2.1.0, which I'm linking here: https://evilmadscience.s3.amazonaws.com/wiki/watercolor/software/RoboPaint.app.zip

Would one or two folks be willing to test and see if this works? (If so, I'll replace the download in the release with this one.)

lalo, 2018-10-20:
@oskay does not work on my mac os 10.13.6, same error

oskay, 2018-10-20:
@lalo Can you please say which error you are getting? We tested downloading on a different computer here and didn't have an issue.

oskay, 2018-10-20:
Here's an updated snapshot to try:
https://evilmadscience.s3.amazonaws.com/wiki/watercolor/software/RoboPaint.zip

lalo, 2018-10-20:
It loads but then has an initialization error, thanks for debugging all this @oskay. Is there something I need to install before hand? Just want to make sure its not an error on my side.

[image]

```EROFS: read-only file system, open '/private/var/folders/9l/rz1znw6x1s316tr5rmwhqvv80000gn/T/AppTranslocation/C28990B8-EFC9-4F76-88D8-D3514DD87D97/d/RoboPaint.app/Contents/Resources/app/node_modules/cncserver/config.ini'

Error: EROFS: read-only file system, open '/private/var/folders/9l/rz1znw6x1s316tr5rmwhqvv80000gn/T/AppTranslocation/C28990B8-EFC9-4F76-88D8-D3514DD87D97/d/RoboPaint.app/Contents/Resources/app/node_modules/cncserver/config.ini'
    at Object.fs.openSync (fs.js:558:18)
    at Object.module.(anonymous function) [as openSync] (ELECTRON_ASAR.js:173:20)
    at Object.fs.writeFileSync (fs.js:1233:33)
    at exports.File.File.saveSync (/private/var/folders/9l/rz1znw6x1s316tr5rmwhqvv80000gn/T/AppTranslocation/C28990B8-EFC9-4F76-88D8-D3514DD87D97/d/RoboPaint.app/Contents/Resources/app/node_modules/nconf/lib/nconf/stores/file.js:93:6)
    at saveStoreSync (/private/var/folders/9l/rz1znw6x1s316tr5rmwhqvv80000gn/T/AppTranslocation/C28990B8-EFC9-4F76-88D8-D3514DD87D97/d/RoboPaint.app/Contents/Resources/app/node_modules/nconf/lib/nconf/provider.js:479:23)
    at Array.reduce (native)
    at exports.Provider.Provider.save (/private/var/folders/9l/rz1znw6x1s316tr5rmwhqvv80000gn/T/AppTranslocation/C28990B8-EFC9-4F76-88D8-D3514DD87D97/d/RoboPaint.app/Contents/Resources/app/node_modules/nconf/lib/nconf/provider.js:521:31)
    at /private/var/folders/9l/rz1znw6x1s316tr5rmwhqvv80000gn/T/AppTranslocation/C28990B8-EFC9-4F76-88D8-D3514DD87D97/d/RoboPaint.app/Contents/Resources/app/node_modules/cncserver/src/cncserver.settings.js:49:25
    at /private/var/folders/9l/rz1znw6x1s316tr5rmwhqvv80000gn/T/AppTranslocation/C28990B8-EFC9-4F76-88D8-D3514DD87D97/d/RoboPaint.app/Contents/Resources/app/node_modules/nconf/lib/nconf/provider.js:406:32
    at /private/var/folders/9l/rz1znw6x1s316tr5rmwhqvv80000gn/T/AppTranslocation/C28990B8-EFC9-4F76-88D8-D3514DD87D97/d/RoboPaint.app/Contents/Resources/app/node_modules/async/lib/async.js:52:16```

oskay, 2018-10-20:
@techninja, any idea what this one is about?

Also, if you can give me a copy that _has  not_ been signed, it may be easier for me to conduct the signing.  I can also share my signing script with you-- if there's any chance that it would help to sign from the CLI rather than automatically.

lalo, 2018-10-20:
Looks like its trying to write something without the permission to do it, possibly related to not having permissions on the /private/ folder.

oskay, 2018-10-20:
I've tested it here on computers running 10.11.6, 10.13.5, and 10.14, and it appears to be working correctly on each of them.

I also don't have write access on /private/, so that's not it (or at least, not the root cause of the problem).

lalo, 2018-10-20:
@jrheard can you test with the latest binary that @oskay shared? I'm know curious if its something on my side. 

Also Oskay, do you have the key somewhere on that computer that the app is able to function correctly because it has the developer or debug stuff installed also so it can figure things out automatically? Maybe try with another user that doesn't have access to wherever you develop the app.

oskay, 2018-10-20:
I don't have debug tools for RoboPaint; I'm not the author of it, and don't know much about how it is built. 

The three computers that I tested on are my own (yes, development) computer that I use to sign things, as well as two non-developer computers, with different OS versions, neither of which has any access to signing tools.

jrheard, 2018-10-20:
same behavior on my machine, big red error box that says

```
EROFS: read-only file system, open '/private/var/folders/zl/bh7pbyz95rg7pmcvcc_f_kdm0000gn/T/AppTranslocation/CB578BF1-A6E6-4BA0-9703-ED351A3144FF/d/RoboPaint 2.app/Contents/Resources/app/node_modules/cncserver/config.ini'

Error: EROFS: read-only file system, open '/private/var/folders/zl/bh7pbyz95rg7pmcvcc_f_kdm0000gn/T/AppTranslocation/CB578BF1-A6E6-4BA0-9703-ED351A3144FF/d/RoboPaint 2.app/Contents/Resources/app/node_modules/cncserver/config.ini'
    at Object.fs.openSync (fs.js:558:18)
    at Object.module.(anonymous function) [as openSync] (ELECTRON_ASAR.js:173:20)
    at Object.fs.writeFileSync (fs.js:1233:33)
    at exports.File.File.saveSync (/private/var/folders/zl/bh7pbyz95rg7pmcvcc_f_kdm0000gn/T/AppTranslocation/CB578BF1-A6E6-4BA0-9703-ED351A3144FF/d/RoboPaint 2.app/Contents/Resources/app/node_modules/nconf/lib/nconf/stores/file.js:93:6)
    at saveStoreSync (/private/var/folders/zl/bh7pbyz95rg7pmcvcc_f_kdm0000gn/T/AppTranslocation/CB578BF1-A6E6-4BA0-9703-ED351A3144FF/d/RoboPaint 2.app/Contents/Resources/app/node_modules/nconf/lib/nconf/provider.js:479:23)
    at Array.reduce (native)
    at exports.Provider.Provider.save (/private/var/folders/zl/bh7pbyz95rg7pmcvcc_f_kdm0000gn/T/AppTranslocation/CB578BF1-A6E6-4BA0-9703-ED351A3144FF/d/RoboPaint 2.app/Contents/Resources/app/node_modules/nconf/lib/nconf/provider.js:521:31)
    at /private/var/folders/zl/bh7pbyz95rg7pmcvcc_f_kdm0000gn/T/AppTranslocation/CB578BF1-A6E6-4BA0-9703-ED351A3144FF/d/RoboPaint 2.app/Contents/Resources/app/node_modules/cncserver/src/cncserver.settings.js:49:25
    at /private/var/folders/zl/bh7pbyz95rg7pmcvcc_f_kdm0000gn/T/AppTranslocation/CB578BF1-A6E6-4BA0-9703-ED351A3144FF/d/RoboPaint 2.app/Contents/Resources/app/node_modules/nconf/lib/nconf/provider.js:406:32
    at /private/var/folders/zl/bh7pbyz95rg7pmcvcc_f_kdm0000gn/T/AppTranslocation/CB578BF1-A6E6-4BA0-9703-ED351A3144FF/d/RoboPaint 2.app/Contents/Resources/app/node_modules/async/lib/async.js:52:16
```

oskay, 2018-10-20:
@jrheard and @lalo:
Did you copy RoboPaint.app to /Applications before launching?

lalo, 2018-10-20:
@oskay you are correct. The error showed when launched from not /Applications. It launches correctly in /Applications folder. Sorry :\ and thanks for helping out!

oskay, 2018-10-20:
Whoa! That was a _long shot_! @jrheard ?

jrheard, 2018-10-21:
Same, I was running it from downloads - afk until tomorrow so can’t try
from applications today, but I’m sure it was the same issue!

oskay, 2018-10-21:
Please let us know how it turns out when you have a chance. Thanks!

oskay, 2018-11-28:
I've updated  the release with this version.

closed 2018-11-28 by oskay