evil-mad/robopaint #302: Recent issues with connecting

oskay, 2020-01-17:
Several WCB users have sent us recent reports of inability to connect via RoboPaint on Windows.  Apparently running older versions of RoboPaint has allowed some (but not all) users to connect.

One user reports a "uncaught TypeError; this.log is not a function" error after the general failure to connect message.

tomalley4, 2021-11-16:
The  "uncaught TypeError; this.log is not a function" error happened to me on my Windows 10 PC 2 years ago and again today. Installing v0.9.5 (a release I know worked in the past) allows me to connect. I wonder if it has to do with where Robopaint installs... with the latest release I was having trouble finding the location I usually add the axidraw.ini file and finally found it under my username instead of Program Files x86.

nickshl, 2024-03-01:
I could not connect. This thread: https://forum.evilmadscientist.com/discussion/629/unable-to-connect
have some useful information. I was able to connect board using PuTTY, perform the R command and get "OK", I also ran SM,100,100,100 which appeared to move the steppers properly. I also enabled  "developer mode" within RoboPaint and got:
```
Available Serial ports: COM5
No matching serial ports detected
```

PervasiveUrbanism, 2024-05-06:
Hi there,

I have the same issue with robopaint. I open the app and I get this message: "Uncaught TypeError: this.log is not a function". I quite don't understand what I can do?! Help would be great!  Thank you very much.